opentx/opentx #3665: AR9X serial protocol

modraude, 2016-07-25:
im using opentx on my ar9x and ar universal because is more rational and clean than ersky9(my opinion), but really i miss the feature of serial output for multiprotocol module. i hope it will be considered in the future.

schwabe, 2016-07-25:
@modraude It is already in next and for master see pull request #3656.

schwabe, 2016-07-25:
Sorry, read the comment to quick. You probably talking about the 9x. The big thing missing for 9x is the UI implementation.

schwabe, 2016-07-26:
I am looking into implement the UI for 9x but I cannot test it. @modraude are you willing to test a beta version?

modraude, 2016-07-27:
ok i can test it reading output with protocol analizer  till i receive the multiprotocol module,

modraude, 2016-07-27:
i just check that i have the material to build a multiprotcol module with cypress

schwabe, 2016-07-27:
@modraude that your time with testing, we can also wait until your module arrives. But giving it a quick look if it looks reasonable should not hurt.

I pushed the ui implementation (with a few other things, need to cleanup before makeing a pull request) to schwabe/multimode9x. I also built a binary from that here; http://plai.de/.tmp/opentx-ar9x-multi22.bin.

I can backport the changes to 2.1 later but would prefer to get it working on 2.2 first. If you try the code backup your eeprom first.

Telemetry might work if I got the serial initialisation for the AVR ARM chip right.

modraude, 2016-08-07:
i test the new firmware. i had this problem, it seems that radio has some audio and lcd ligth glitches an after some glitches radio reboot by itself.
i try to change some menu before reboot of radio thinkink may be i can stop this behaviour but i dint reach . i hope this poor info can help you.
ciao eduardo

schwabe, 2016-08-08:
hm, those crashes and problems are probably unrelated to thee multi code and a result of not testing next code on the ar9x or other 9x arm platforms. 

you could try to install a "normal" next version to narrow the problem down. Unfortunately, I cannot look at the problem myself since I am missing the hardware to debug it

schwabe, 2016-09-12:
Are the 2.2 opentx binaries still crashing for you? (without multi?)

schwabe, 2017-01-13:
Multi ui on ARM 9X is working fine (reports x7). I am closing this issue. For more specific multi+arm 9x please open new tickets.

closed 2017-01-13 by schwabe