opentx/opentx #3667: DCR.  PRI-LOW.  Show/Hide menu tabs

StevenSeim, 2016-07-27:
Hello.  With permission, I would like to submit a low priority design change request. (DCR)

Within Companion, on opening a model, grant the option to show/hide the menu tabs through a settings icon (say, a gear) on the far right of the master frame.  The Setup tab would remain fixed.

Result:  

1. Users that do not fly helicopters will have the ability to hide that tab from view if wanted.
2. Users with models not complex can hide what they don't leverage from the application.
3. Streamlining and more rapid accessibility. 

Many thanks to you.
Steven

3djc, 2016-07-27:
Does this duplicate https://github.com/opentx/opentx/issues/3491 ?

kilrah, 2016-07-27:
Yep

closed 2016-07-27 by kilrah

StevenSeim, 2016-07-27:
It may.  

 

I had not seen 3941.  On reading, it sort of defines the symptoms and problem, but does not clearly define the solution.

 

If it makes the same point, please roll my issue into 3491 and close it out.

 

Many thanks for responding quickly.

 

 

Steven

 <https://www.facebook.com/steven.j.seim> Facebook

 <http://www.cybermodelle.com/> Web

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2016.0.7688 / Virus Database: 4627/12676 - Release Date: 07/24/16

kilrah, 2016-07-27:
The solution is pretty clearly defined and would take care of your issue, but the current state of the discussion has objections to implementing that altogether.

StevenSeim, 2016-07-27:
Understood.

 

It is a DCR, and low priority at that.  Usability is not adversely affected nor adversely hindered.

 

I would suggest though that it would be an aid for folks that do not leverage all the menu tabs, given the option to reveal/hide all tabs at will.

 

Having the ability to show only the menu tabs most commonly used might aid in porting the application to a mobile platform perhaps?

 

Again, many thanks for responding promptly.

 

 

Steven

 <https://www.facebook.com/steven.j.seim> Facebook

 <http://www.cybermodelle.com/> Web

kilrah, 2016-07-27:
Please read the other issue for details about why that couild be a bad idea.