opentx/opentx #3668: Compatibility with "new" Turnigy TGY 9x

ekke2ekke, 2016-07-28:
Hi,

Opened my Turnigy 9x (bought last month from Hobbyking) and it doesn't have Atmel mcu. It's manufactured by Green and might be Atmel compatible. Any ideas if it can be flashed with Opentx?

kilrah, 2016-07-28:
Any detailed photos?

ekke2ekke, 2016-07-28:
https://www.dropbox.com/s/rq38j9fdwiuwziq/DSC_0812.JPG?dl=0

Couldn't do better with my phone, but might be enought?

schwabe, 2016-07-28:
That seem to be a chip from a Chinese company that makes AVR clone chips (http://www.lgt-semi.com/?lang=en) 

Bt at least from the website they don't make any chip with a larger flash than 8k (http://www.lgt-semi.com/?page_id=5297&lang=en)

The part number might be ED040501-H16D http://elcodis.com/parts/6269600/ED040501-H16D.html

kilrah, 2016-07-28:
The PCB looks identical (although it has silkscreen, it used not to) and I highly doubt FlySky would have touched their dinosaur of a firmware, so IMO that chip is highly likely a 100% compatible clone.

ekke2ekke, 2016-07-28:
Well, not sure if I want to try because it's my only transmitter before Horus comes out.. So maybe 2019? ;)

(I have Arduino usb2serial that has isp-header..)

kilrah, 2016-07-29:
OK, closing this then.

closed 2016-07-29 by kilrah