opentx/opentx #3673: [Bug][Radio][2.1.8] Bug with Spektrum Hack Module

LapinFou, 2016-07-31:
Hi folks,
When creating a new model with the hack module, choosing "DSM2 DSMX" will coerce the channel to CH1-**8**.
It should be CH1-**6**.
I can reproduce the problem on my Taranis and also with the simulator.

LapinFou, 2016-08-11:
Will this "cosmetic" bug also be fixes in the OTX 2.1 branch?

schwabe, 2016-08-11:
@LapinFou It is already fixed in 2.1. And I called it cosmetic because the only thing that the commits changes is what is shown in the. The DSM2 code does not evaluate the channel count always sends six channels to the module

closed 2016-08-11 by schwabe

LapinFou, 2016-08-11:
Thanks for the clarification. :smile: 
Since it was tagged for the Milestone ***OpenTX 2.2.X*** I thought it will not be fixed for ***OpenTX 2.1.X***.