opentx/opentx #3685: Lua debugging requests – part 1

pau33, 2016-08-03:
I worked a lot the last time on writing complex mix scripts in lua for the taranis. Debugging of these scripts in OpenTx is not the easiest thing. So I have 3 request for making debugging easier. The first is the possibility to apply a filter to the debug console output (PF6) in companion. For example in a way, that only lines beginning with a defined string (for ex. „lua“) are shown. So I have partly the ability to look at my print commands (starting with „lua...“) or at the lua script error lines. Another solution would be to print the lua statements additionally into a second window. Also the ability to delete all content in the debug console output would help.

projectkk2glider, 2017-01-14:
Nice ideas.

mpaperno, 2017-01-14:
Filter is a great idea and should be pretty easy to add.  The file I/O logging can get pretty verbose indeed.  Also the GC messages can flood the screen on more complex scripts.

I wonder if filtering messages at a higher level would be useful in addition or instead (perhaps at simulator program interface level).  I realize there are some build-time options to control that now, but those aren't helpful for typical use.

Also on my list is getting the debug info window to remember its size and position instead of obscuring the sim each time.

-Max

mpaperno, 2017-02-06:
Closed via #4385.

closed 2017-02-06 by mpaperno