opentx/opentx #3686: Lua debugging requests – part 2

pau33, 2016-08-03:
A lot of lua debugging can be done in the companion simulator, but not all. At the end you need the radio for the rest. I wrote a debugging function, which writes my debugging output to a file, conditioned by a switch. I can spend the code for the lua Manual, if there is a request for. Things would be easier and also enhanced in this field, if there is for example an additional option „d“ or „debug“ for the lua io.open command in a way, that all lua print commands and also the lua script errors an the reasons for forcefully closing would be written to this file, if opened it in that way.

projectkk2glider, 2017-01-14:
Perhaps now that we have Lua compiler active, we need to add a new radio setting: `Lua Debug Mode`. 

When turned `ON`:
 * the Lua compiler is disabled
 * Lua debugging prints got to a special file on the SD card (while also going to debug console if active like now)

When turned `OFF` (this would be normal for majority of users):
 * Lua compiler enabled
 * Lua debugging output to file disabled

mpaperno, 2017-01-14:
I like the option of debugging to a file on the radio.  It can definitely be useful to see what's going on in the actual radio.  

Although, can one see Lua errors now if using a terminal with the new USB VCP option (and w/out a DEBUG build)?  I tried it once but now can't remember... went back to using dedicated serial port on X9D.

Anyway, it is easy (change one argument) to tell the pre-compiler function which version of a script to load (source or binary), so that is no problem to disable (if we had a runtime setting or whatever).  Also the new [loadScript()](https://github.com/opentx/opentx/blob/next/radio/src/lua/api_general.cpp#L1141) API function takes the same optional argument to determine which version to load, so this is already controllable on a per-script level.  If you're loading scripts with `io`, then the compiler feature doesn't even come into play.

-Max

mpaperno, 2017-01-14:
BTW, compiler not active by default on firmware builds yet.  Though I think it should be.  😉

closed 2020-04-06 by stale[bot]