opentx/opentx #3687: Lua debugging requests – part 3

pau33, 2016-08-03:
The last request is an easy one. In the custom script page on the Taranis there is a „Name“ field for each lua script. I can set this manually on the radio. A good thing would be, if there is a way, if I could write to this field from within lua (for example setLUAname and also getLUAname). Why? Mostly I use the same .lua name for several versions (alpha, beta, release candidate) of the lua file. So I don't have to set my script inputs always new on the radio. If I could write to the Name field form within lua, I have the control on the radio, that I use the right script version. Also scripts with the same function but different parameters for different models could be easily marked in this field form the lua script itself.
Last side question: What does the percentage number on the mix screen right to every lua script tell me exactly and what is the highest possible amount of bytes, the mix scripts can have on the radio (bytes in the line above the lua mix scripts)?

kilrah, 2016-11-22:
Closed by OP request

closed 2016-11-22 by kilrah

lshems, 2016-11-22:
What is an OP request?

3djc, 2016-11-22:
Original Poster