opentx/opentx #3688: FEATURE request (telemetry data in variables A1, A2 and other discoverable) (9xr pro)

njozsef, 2016-08-04:
Hi developers.

My fpv racer setting hover point is stick center. (weight and offset). This good is first 1 min.
if voltage dropping is need shifting hover point. 
Need an mathematical fuction. Example GV1=16.5-A1*0.2

Thank you.

Jozsef

kilrah, 2016-08-04:
You can use a telemetry value in an input, and use that as a correction curve in the mixer.

closed 2016-08-04 by kilrah

njozsef, 2016-08-04:
I revisioned my request. 
Please simply insert A1 and A2 and Timers the source paramete list. Like P1 p2 p3

Please see picture:
https://drive.google.com/file/d/0B1tSgWAfLbkBNTRONktaVkZrVDg/view?usp=sharing

kilrah, 2016-08-04:
Ah, didn't notice what radio you were using. That is supported on the Taranis but not on 9XR-Pro.