opentx/opentx #3691: Companion 2.2 Crashes (access violation) when Taranis simulator window closed

jtaylor2, 2016-08-05:
Companion 2.2 Downloaded companion-windows-2.2.0N345.exe nightly and also latest next compiled on Linux openSUSE 42.1 and Windows with VS 2015.

Steps to reproduce:
1. Run Companion 2.2 - Radio type Taranis
2. Open simulator window either from Model List screen or Edit Model screen
3. Close simulator window either by menu-close or clicking corner x
4. Companion crashes with a Access Violation on Windows or Segmentation Fault on Linux

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcd06e700 (LWP 2235)]
0x00007fffdd5be210 in boardOff ()
    at /root/frsky/opentx/git/opentx/radio/src/targets/taranis/board.cpp:224
224       SysTick->CTRL = 0; // turn off systick
```
I have fixed this locally with:

```
diff --git a/radio/src/targets/taranis/board.cpp b/radio/src/targets/taranis/board.cpp
index 8488b9c..4f95748 100644
--- a/radio/src/targets/taranis/board.cpp
+++ b/radio/src/targets/taranis/board.cpp
@@ -221,7 +221,9 @@ void boardOff()
 #endif
 
   lcdOff();
+#if !defined(SIMU)^M
   SysTick->CTRL = 0; // turn off systick
+#endif^M
   pwrOff();
 }
```

I'm reluctant to do a pull request because I'm not sure this is the best way to fix it, but I can if it looks like the right way.

3djc, 2016-08-05:
It is indeed an issue on our to-do list

3djc, 2016-08-18:
Fixed in next

closed 2016-08-18 by 3djc

voyager9, 2016-09-03:
This is not Fixed. It still crashes by closing the simu window