opentx/opentx #3694: Compatibility with "new" Turnigy TGY 9x (previous #3668)

assa1ssin, 2016-08-09:
I found the same issue, but it was closed (https://github.com/opentx/opentx/issues/3668)
I got the radio from the same batch - it has atmel processor clone - ED040501-H16D
looks like it is clone of atmega 128 - I was not able to flash it with m64 opentx FW, but it was possible with m128.... flashing firmware was without problems, but it is not working correctly - It reboots sometimes, it shows the exclamation mark etc. I tried the "trim mod" - digital grounds for the trims, but it didnt help. Now after some time one of the trims seems like pressed all the time, but it is not, it does the same even if I remove the wires (except ground) from the trim PCBs....
So for me, this new radio is useless, maybe the original turnigy FW is working, but without open tx its regular cheap TX :(

schwabe, 2016-08-09:
This is a duplicate of #3668. Sounds like the atmega 128 clone is compatible enough to run OpenTX without problems.

While that sucks, there not much this project can as there virtually nothing known about the clone 128 and its differences to a real 128.

bsongis, 2016-08-09:
Is the stack remaining size OK?

assa1ssin, 2016-08-10:
It shows free stack - 989b, after my experiments with grounds for the horizontal trims, after few seconds the TX starts to beep and it looks like the trim is pressed (vertical one) nothing helps, grounds are back how they was in stock and I cant see any mistake. I understand that there is not much to do.

bsongis, 2016-08-10:
Would it help if I sent you a firmware which disables the trims to test if
the problem is there or somewhere else? If yes please send me a direct
E-mail

assa1ssin, 2016-08-10:
bsongis: yes, this would be fine, you can use this one: assassin_tl(at)centrum.cz
what exactly means this exclamation mark? it shows after few seconds of "walking through" menus and then it reboots. I didnt find the reason for reboot, It happened during the flight twice, no trims or buttons were touched...

guilhermebjr, 2016-08-25:
Same problem... Waiting for...

bsongis, 2016-08-25:
I will be off tomorrow (friday). Please send me a message if I forget you again. French hours!

bsongis, 2016-08-27:
Here:
http://downloads.open-tx.org/2.2/tests/

assa1ssin, 2016-08-31:
Thanks, but when I try to upload FW it says that it is expecting atmega64 processor, but inside is 128 clone - do I have to override this check, or it is neccesary to modify different FW from Your side?
Thanks, Jan

bsongis, 2016-08-31:
I didn't know it was a '128 ...

assa1ssin, 2016-09-19:
I wrote it in the first post....  Is it possible to modify the FW for 128 too? Thanks

assa1ssin, 2016-12-06:
problem solved - resoldered atmega instead of chinese clone and now it works without problems, so I can say, that the clone processor is the reason of the malfunction.

projectkk2glider, 2017-01-14:
See the official statement from Flysky here http://openrcforums.com/forum/viewtopic.php?f=126&t=9387&start=60#p122326. They basically say that GREEN processor is the same and should work. The problem is in the flashing.

closed 2017-01-14 by projectkk2glider