opentx/opentx #3695: DIY Spektrum modules built using recent MLP4DSM module revisions don't work.

Fheilmann, 2016-08-10:
DIY Spektrum modules built using newer MLP4DSM modules (identified by a blue color dot on the main IC and "DSMX" text on the remote) don't work with the current version of OpenTX (on a Taranis Plus). The reason for this is that the baud rate has changed for these modules from the standard 125k baud that was used in earlier modules.

According to [this rcgroups post](http://www.rcgroups.com/forums/showpost.php?p=25961890&postcount=14) the new modules run at 133k baud. 

To me, the modification that makes the most sense to incorporate these newer modules while keeping compatibility with the older modules (DX4e DX5e, HP6DSM) is to make the baud rate configurable in the model setup and set the baud rate in [DSM2_Init()](https://github.com/opentx/opentx/blob/dbd8abbfe8343b5d7f542304d47e232140307b95/radio/src/pulses/pulses_avr.cpp#L449) accordingly. But I am a OpenTX coding newbee at best, so I don't know if this is the only necessary change.

schwabe, 2016-08-10:
I have a fairly recent MLP4DSM at home. I can take a look at it when I have time.

Btw. the DSM2_Init function you found is only for AVR. As I understand you have a Taranis.

https://github.com/opentx/opentx/blob/dbd8abbfe8343b5d7f542304d47e232140307b95/radio/src/pulses/dsm2_arm.cpp

Changing the BITLEN_DSM2 define to 15 instead of 8*2 might be enough to get that stuff working. If you want I can generate a 2.1.8 binary for you to do that quick&dirty test. And then implement a proper fix if that works.

Fheilmann, 2016-08-10:
@schwabe thanks a lot for assisting with this. I'll try to build a version myself as an exercise, if I don't succeed I'll get back to you.

Fheilmann, 2016-08-15:
Some more thoughts. I'll test with 133k baud today (i was set back over the weekend so I'm a little late on this one) but according to the blog post I linked, 131k or 138k may be necessary. The issue here is that [_send_1()](https://github.com/opentx/opentx/blob/dbd8abbfe8343b5d7f542304d47e232140307b95/radio/src/pulses/dsm2_arm.cpp#L55) expects a `uint8_t` as duration, which works fine for 125k and 133k but not for 131k or 138k which would require 15.2 and 14.4 to be passed to the function.

Is it even possible to change the underlying functionality to allow for these non-integer timings to be used?

schwabe, 2016-08-15:
@Fheilmann no. But the underlying frequency could be changed. But that blog post seem to be only one suggesting 131k or 138k, the other source of rcgroups between of between 131-138k.

Fheilmann, 2016-08-15:
Ok so I tried with ~133k and it didn't work. I remember somewhere else that the newer modules may also have higher voltage requirements than 3.3V. But that would take a little more effort to test.

schwabe, 2016-08-16:
My logic analyzer likes the spektrum signal at 133 8N0.5 or at 137k 8N1, 138k already gives me framing errors:
[image]

and Taranis with 15 really outputs 133k 8N1:

[image]

And the module is still 3,3V.

I am not familar with the code that configuration  the gpio to change its base frequency to something that results in something usable for 137k. My attempts resulted in a watchdog reset :/

Fheilmann, 2016-08-16:
Nice findings! Maybe someone that has knowledge of the gpio configuration can chime in?

schwabe, 2016-09-05:
As a crude hack you can try to modify taranis/pulses_driver.cpp:

And change that 
```
  EXTMODULE_TIMER->PSC = EXTMODULE_TIMER_FREQ / 2000000 - 1; // 0.5uS (2Mhz)
  EXTMODULE_TIMER->ARR = 44000; // 22mS
```
into 
```
  EXTMODULE_TIMER->PSC = EXTMODULE_TIMER_FREQ / 2192000 - 1; // 0.456uS (2Mhz)
  EXTMODULE_TIMER->ARR = 48224; // 22mS

```

and see if that works (untested)

petrochen, 2016-09-05:
Same thing with new X10EMTX module. Last changes not working for me.

JeremyJass, 2016-10-02:
Chiming in to verify the issue.  Signal levels from the Osh Park DIY mod board were a bit of a mess (though probably would have worked fine since loads of people use it) and what I first assumed to be the culprit, but I now have the signal from my new Taranis Plus (running 2.0.20, or 2.1.9) fed through a dual stage open collector with a separate voltage supply to get a clean 0-3.3V signal with sharp edges and still no dice.

I left the logic analyzer at work, so I'm just looking at an o-scope, but the signal the new MLP4DSM I got with my Inductrix was feeding to the X10EMTX (with blue dot) was fundamentally different than the Taranis DSMX signal.

The MLP4DSM serial data packets are about 1.06ms long and made of up high's and lows that are always even multiples of just under 8us.  The Taranis serial data packets are about 1.22ms long and have pulses as short as 4.4us, but then jump to 12.4us with no intermediate length pulses.  Maybe the protocol only requires the signal to be high at the start of each bit or something, but the Taranis signal looks weird.

Finally, the channel order in the serial data from an MLP4DSM in Mode 1 and a Taranis DSM2/DSMX output in Mode 1 are different.  I'll try to research some more and maybe hook it up to the logic analyzer next week.  My ACCST receiver and micro brushless flight controller should be shipping from BangGood next week as well, so my motivation to solve this may unfortunately be short lived :D

schwabe, 2016-10-02:
My Inductrix MLP4DSM signal looked almost identical to the one of OpenTX apart from the baudrate.

JeremyJass, 2016-10-02:
So the signals are pretty similar, and with a different baud rate my logic analyzer was able to decode both.  It was able to parse the MLP4DSM signal at a range of frequencies around 136 kHz and I'm not sure exactly how to work back from the signal to the sample rate to know the exact "best" fit.  I've attached raw outputs from the Taranis (2.1.9, DSM2/DSMX) and the MLP4DSM as csv's in case that's helpful.

I'll try to set up the toolchain to start compiling and testing code to see if I can't find a solution this week.  Probably need a new option on the menu to allow changing the frequency since all the people with older Spectrify modules will need to have the older 125kHz rate.

Taranis output analyzed at 125 kHz:
[image]

New MLP4DSM analyzed at 136kHz:
[image]

[[raw_csv_outputs.zip](https://github.com/opentx/opentx/files/504992/raw_csv_outputs.zip)]
Taken at 1MS/s sample rate, but I can run it again at a higher rate if that would be helpful.

FHeilmann, 2017-01-05:
Any updates on this? I guess with all the frsky compatible microquad boards surfacing this is not as important anymore.

JeremyJass, 2017-01-05:
Yeah, I've given up and my Inductrix will just have to make do with the toy grade controller it came with.  There are frsky compatible radios in just about any form factor, so I'm just making sure anything new I build or buy speaks frsky.

It is pretty clear the new MLP4DSMs modules have a different clock frequency then the older ones, but changing the clock frequency on the serial output of the Taranis isn't easy or very straight-forward.

kilrah, 2017-01-05:
+1, I did have a hack DSM module but even if it still worked I replaced it with a multimodule anyway for convenience since that does everything, so no more module swaps.

JBFUK, 2017-03-22:
Is this why my module won't bind.  I did get it to bind eventually but this was the result:  https://www.youtube.com/watch?v=3CpK_mm0_kA

FHeilmann, 2017-07-05:
Since this has not received any additional attention and multiprotocol modules are widely available now I'm closing this.

closed 2017-07-05 by FHeilmann