opentx/opentx #3697: Warnu users to upgrade sound files

mpjf01, 2016-08-12:
Barely a day passes without a user asking on a forum for help because his voice readouts are wrong after an upgrade from 2.0.x to 2.1.x. It would be useful if users could be warned at the point of doing the upgrade that they need to attend to their sound files and given a link to the new versions. It won't work for everyone because some people won't read instructions like this, but it may help.

I understand that further changes are happening for 2.2 so the issue will arise again.

I'm not sure how Companion can find out, perhaps by looking for previous installations on the computer. Many other applications manage it so there might be a standard way.

If that is not possible, Companion seems to know when users have upgraded from a previous version when they attempt to load an old eepe so the warning and link could be given then.

3djc, 2016-08-12:
There is bigger need than just sound. I'm currently thiinking that the radio should check a minimum sd 'version'

bsongis, 2016-08-12:
Excellent idea, I would like to include this feature in 2.2, @3djc would
you like to take it? I am about to push a giant change for X7D and Sky9X
and virtual inputs ...

<https://github.com/opentx/opentx/issues/3697#issuecomment-239371093>, or mute
the thread

3djc, 2016-08-12:
Yes, I'll take that

kilrah, 2016-08-12:
The radio should check for presence of all important files on each startup, not just on upgrade to catch when people mess up with their card.
On ARM we could also do away with the numbered files (besides numbers...) and give them meaningful names...

3djc, 2016-08-12:
Then I propose the following:
- we manage sound as we do with all other sd cards files in github. That will ensure consistency in versions and allow better community improvement management.
- as part of the compile process, we generate a list of files that need to be present (ie all of radio/scdard/platform) and feed that to the firmware sources
- at radio startup, those file are checked and a summary is displayed should files be missing

jtaylor2, 2016-08-12:
The current (2.0-2.1 sound pack not upgraded) problem isn't just that they are missing, but that they exist and contain the wrong sounds.  Checking that they exist won't fix that problem.  Checking SUMS would, but then people couldn't use their own packs, and I can't imaging anyone would want to give up using the Amber pack.

Putting a version file on the card and having the radio check it at startup and display a message if it isn't correct seems like the best answer.  It would warn people that they are using the wrong pack, but still allow for custom sound packs.

kilrah, 2016-08-12:
>The current (2.0-2.1 sound pack not upgraded) problem isn't just that they are missing, but that they exist and contain the wrong sounds. 

Some do, but some are missing too. The warning could be "you have files X, Y and Z missing so your SD card files may be outdated, check them all anyway"

3djc, 2016-08-12:
So maybe we should:
- get away with numbered sound file, and move to english named files
- we build an index of all the files that need to be there (required audio + radio/sdacrd/platform content)
- at startup, we check all files in that index

Teracis, 2016-08-31:
Hi all,

Just a thought to add to your warning about missing files, in order to have the best of both worlds, maybe  you could set it up so that it checks on start up for the files you require, but have an option in a menu somewhere that allows this check to be disabled.

That would take care of all the new users as they would get the message, then allow for those who understand what is going on to still continue doing what they are doing (custom sound packs e.t.c.) without an interruption every time.
That way you could go with the checksum idea and not actually break it for custom packs.
Even maybe two options, standard checks sums and the index, menu option for turning both checks off independently.

Just a thought, I have no idea how to actually implement this stuff or how much extra work I've suggested for you guys, just brainstorming with your ideas. I was led here searching for changes made to the sound files after I saw someone ask about the amber sound pack for 2.2 on RC Groups.

Cheers
Tim

bsongis, 2016-08-31:
Done by @3djc thanks a lot!

closed 2016-08-31 by bsongis

RCdiy, 2016-09-08:
There were a number of suggestions. What has been implemented?

kilrah, 2016-09-08:
There's an opentx.sdcard.version file on the root of the SD card that contains a version number. If we make a change in the requirements we increment the number the firmware expects, and a mismatch will throw a warning.