opentx/opentx #3698: Unknown PCB When cmake'ing 2.2

davidbitton, 2016-08-14:
I'm trying to build 2.2 (next) on my Mac. Here is the results:

```
bittondb:build/ (next) $ cmake -DPCB=XD9+ -DHELI=YES -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt5/5.6.1-1 ..                                          [15:31:38]
-- The C compiler identification is AppleClang 7.3.0.7030031
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenTX 2.2.0
-- Looking for SDL
SDL not found! Joysticks support will be disabled. Simulator audio will not work.
-- Looking for FOX
-- Could NOT find FOX (missing:  FOX_INCLUDE_DIR FOX_LIBRARY)
Libfox not found, simu target will not be available
-- The ASM compiler identification is Clang
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
CMake Error at radio/src/CMakeLists.txt:99 (message):
  Unknown PCB 'XD9+'

-- Configuring incomplete, errors occurred!
See also "/Users/bittondb/src/opentx/build/CMakeFiles/CMakeOutput.log".
```
This happens with _any_ board type I select. Thanks!

jtaylor2, 2016-08-14:
It's PCB=X9D+ and PCB=X9D see radio/src/CmakeLists.txt lines 67-75

davidbitton, 2016-08-14:
It's working. I had to install fox and sdl via brew. Thanks!

closed 2016-08-15 by kilrah