opentx/opentx #3699: Unable To Build 2.2 'next' Companion; Missing Lupdate

davidbitton, 2016-08-14:
I have qt55 installed on my Mac, and cmake is failing with this

```
bittondb:build/ (next✗) $ cmake -G 'Unix Makefiles' ../src                                                                  [19:52:47]
-- Looking for QT lupdate
CMake Warning at CMakeLists.txt:35 (find_package):
  By not providing "FindLupdate.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Lupdate", but
  CMake did not find one.

  Could not find a package configuration file provided by "Lupdate" with any
  of the following names:

    LupdateConfig.cmake
    lupdate-config.cmake

  Add the installation prefix of "Lupdate" to CMAKE_PREFIX_PATH or set
  "Lupdate_DIR" to a directory containing one of the above files.  If
  "Lupdate" provides a separate development package or SDK, be sure it has
  been installed.

QT lupdate not found!
CMake Error at shared/CMakeLists.txt:19 (qt5_wrap_cpp):
  Unknown CMake command "qt5_wrap_cpp".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.6)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/Users/bittondb/src/opentx/companion/build/CMakeFiles/CMakeOutput.log".
bittondb:build/ (next✗) $
```

I've used about every combination of `CMAKE_PREFIX_PATH` and `CMAKE_MODULE_PATH` to no avail. Has anyone come across this while build Companion on OSX? Thanks!

davidbitton, 2016-08-15:
Using the following CMAKE_MODULE_PATH, I at least was able to get a _new_ error output.

```
bittondb:build/ (next✗) $ cmake -DCMAKE_MODULE_PATH=/Users/bittondb/src/opentx/cmake  -G 'Unix Makefiles' ../src            [20:01:21]
-- Looking for QT lupdate
-- /usr/local/Cellar/qt55/5.5.1/bin/lupdate
QT lupdate: /usr/local/Cellar/qt55/5.5.1/bin/lupdate
CMake Error at shared/CMakeLists.txt:19 (qt5_wrap_cpp):
  Unknown CMake command "qt5_wrap_cpp".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.6)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/Users/bittondb/src/opentx/companion/build/CMakeFiles/CMakeOutput.log".
bittondb:build/ (next✗) $
```

Thanks.

davidbitton, 2016-08-15:
My latest command

`cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt55/5.5.1 -DCMAKE_MODULE_PATH=/Users/bittondb/src/opentx/cmake  -G 'Unix Makefiles' ../src`

still isn't working. Ideas?

schwabe, 2016-08-15:
You need to adjust that path to the version you actually have installed via brew.

davidbitton, 2016-08-15:
adjust what component of the path? isn't specifying qt55 via CMAKE_PREFIX_PATH enough?

davidbitton, 2016-08-15:
I added `find_target(Qt5Core)` as the find of Lupdate and then received this

```
bittondb:build/ (next✗) $ cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt55/5.5.1 -DCMAKE_MODULE_PATH=/Users/bittondb/src/opentx/cmake  -G 'Unix Makefiles' ../src                   [22:29:34]
-- Looking for QT lupdate
-- /usr/local/Cellar/qt55/5.5.1/bin/lupdate
QT lupdate: /usr/local/Cellar/qt55/5.5.1/bin/lupdate
CMake Error: File /Users/bittondb/src/opentx/companion/src/companion/src/version.h.in does not exist.
CMake Error at CMakeLists.txt:197 (configure_file):
  configure_file Problem configuring file

CMake Error: File /Users/bittondb/src/opentx/companion/src/companion/src/translations.qrc.in does not exist.
CMake Error at CMakeLists.txt:198 (configure_file):
  configure_file Problem configuring file

CMake Error: File /Users/bittondb/src/opentx/companion/src/companion/src/companion.desktop.in does not exist.
CMake Error at CMakeLists.txt:199 (configure_file):
  configure_file Problem configuring file

CMake Error: File /Users/bittondb/src/opentx/companion/src/companion/src/simulator.desktop.in does not exist.
CMake Error at CMakeLists.txt:200 (configure_file):
  configure_file Problem configuring file

CMake Error at CMakeLists.txt:219 (qt5_add_translation):
  Unknown CMake command "qt5_add_translation".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.6)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/Users/bittondb/src/opentx/companion/build/CMakeFiles/CMakeOutput.log".
bittondb:build/ (next✗) $
```

kilrah, 2016-08-15:
You seem to be pointing cmake to the wrong directory, should be the root of the source code.

3djc, 2016-08-15:
You could also try the docker route, as it is now available for Mac and provide (i believe) linux images support, and the docker dockerfile to compile 2.2 is available here : https://github.com/opentx/opentx/blob/next/radio/util/Dockerfile

schwabe, 2016-08-15:
@davidbitton The example from the wiki needs to fit your current brew qt. E.g. for me it is /usr/local/Cellar/qt5/5.6.1-1 and not 5.5.1 anymore

davidbitton, 2016-08-16:
@3djc but I need to build the OSX version. Has anyone successfully built 2.2 on Mac, and if so, how?

davidbitton, 2016-08-16:
just for shits and giggles, i ran `cmake -G 'Unix Makefiles' .` in the repo root. well, son of a gun, it worked with only a small fix. I found https://github.com/Homebrew/legacy-homebrew/issues/29938#issuecomment-104722532 and just created the necessary symlinks using the 5.5 folder. Then I followed that up with a `make companion` and now I'm running 2.2 Companion.

closed 2016-08-16 by kilrah