opentx/opentx #3703: [Companion 2.1.8N836 Simulator][OpenTx 2.1.8N836] Inputs->Edit->Stick side->NEG, Bug?

483917, 2016-08-17:
Windows 10 pro swedish
Companion 2.1.8N836
Taranis B, Firmware: 
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.8N836.bin"

In ModelEdit->Inputs->Edit->Stick side->NEG, do not behave like StickSide->POS does
Open my demo file: "Neg-switch-bug.eepe/.Zip" 
[Neg-Switch-bug.zip](https://github.com/opentx/opentx/files/422322/Neg-Switch-bug.zip)
 and start Companion Simulator,
in sim-outputs look at Channel 6 = "StickSide POS" , And Channel 7 = "StickSide NEG"
and move switch SA-up , SA-mid , SA-down

Channel 5 is StickSide ALL for referens only, goes from -100 ,0 ,100 as it should
Channel 6 moves from 0 to 100 as it should
Channel 7 do not move from 0 to -100 as it should ?? it dont move at all

This only happens when a "Switch is assigned as Source"
Tested on Companion-sim and on Taranis B ,same behavior

If this is intended behavior ?,then feel free to close this issue
But feels confusing.

3djc, 2016-08-17:
Well there might be a logic flaw in what you are trying to achieve. Since you use a switch, it can have only 3 values, and in your setup, you kinda assume 0 will be both LEFT and RIGHT

schwabe, 2016-08-17:
I looked at the setup. The bug narrows down to SA with stick negative is a constant -100 while for positive it is 0 or 100. 

My bet is that the stick side condition was never meant for switches and somewhere a < 0, <=0 ... condition lurks that is true for pos but not for negative. While stick side is a strange thing for switches, it should be consistant or removed.

483917, 2016-08-17:
Yes i think it should be consistant, StickSide POS 0 to 100 and StickSide NEG -100 to 0
But you decide

kilrah, 2016-08-17:
0 is considered positive, so when your switch goes from -100 to 0 the value is never updated.

<= 0 would make most sense but there's likely a reason for it to be that way, needs careful investigation.

schwabe, 2016-08-17:
What I noticed is that the preview (the curve graph) actually shows values of -100 and 0.

closed 2019-11-15 by stale[bot]