opentx/opentx #3705: Throttle power loss

Bretol, 2016-08-18:
Hello, I don't know if its a OpenTX issue, but I will explain my case:
OpenTX version is 2.1.6

I have a Taranis X9D with a D4R-II receiver.
The flight Controller is a CC3D Atom.
I'm using FlexiPort as PPM input (My Atom board does not receive signal in PIN8 on the MainPort)

After some flights, the drone starts losing power with fully-charged LiPos.
Testing in LibrePilot CGS, the motors seems to maintain stable at max rpm.
Arming the drone with the Taranis and pull throttle stick to the max. it starts at max. rpm (like the LibrePilot test) but losing  gradually RPM's on all the motors (I don't have oscilloscope to check if the radio signal changes)

Please, check this video (with sound): https://vid.me/eczC

I have checked the throttle stick graph and it moves from -100 to +100 proportionally.
I have created a new model with same result.

Any idea?

3djc, 2016-08-18:
Very unlikely to be an OpenTx issue. I have no experience with LibrePilot, but if it allows you to monitor the channel input, that would be a great place to start. If LibrePilot 'sees' the throttle imput correctly, then that means neither your radio or receiver are not the cause

tomkfly, 2016-08-18:
If your channel monitor shows 100 it isn't an open Tx problem, so is not an issue for posting here. Ask on one of the forums.

Bretol, 2016-08-18:
Thank you anyway. :-)

closed 2016-08-18 by 3djc

kilrah, 2016-08-18:
It's actually pretty normal since the motor test in the app bypasses the regulation. Some of it may be due to bad ESC setup/calibration too.