opentx/opentx #3706: Feeding SPort Telemetry into Taranis

marcdornan, 2016-08-20:
Apologies in advance if this is not an appropriate posting here but i am looking for guidance and help. 

I am using [ULRS ](http://www.rcgroups.com/forums/showthread.php?t=2037442)with a Taranis to do long range Mavlink and RC control. I have used a [Teensy Mavlink to FrSky converter ](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.1.-Teensy-and-APM-pixhawk)with X8R Rx to get Mavlink telemetry on my Taranis. What I would really like to know is how can I use a Teensy on the ground side when I am using a UHF module in my Taranis. I gather I an X Port receiver is needed to poll the sensors and transmit the telemetry. There is no easy way to feed S Port into a Taranis on the ground side? If anyone has any ideas I would love some pointers. 

The fallback position is to use an X series receiver to relay PPM and Telemetry back to my Taranis but if there is another way I am eager to find it.

3djc, 2016-08-21:
I'm afraid this is really not the right place to ask that question as it is a bug tracking system. You should consiider forums like http://openrcforums.com/ or http://www.rcgroups.com/

closed 2016-08-21 by 3djc

marcdornan, 2016-08-21:
No worries -- I think I found the answer last night anyway.