opentx/opentx #3708: openTX 2.1.8 - telemetry simulator - replay rate too fast...

nostromo123, 2016-08-21:
...with some of my csv log files.
See thread here -> http://openrcforums.com/forum/viewtopic.php?f=94&t=9121
Did try to upload a screen recording, but it seems, file is too big to be attached here. So I placed it in youtube. https://youtu.be/3snpebY2rDw

nostromo123, 2016-08-21:
Seems, that when time difference between first and second entry in log file is below xxx 1/1000s, that issue shows up.

Checked all of my 21 log files for that.

See attached pdf.
[Unbenannt 1.pdf](https://github.com/opentx/opentx/files/428882/Unbenannt.1.pdf)

jtaylor2, 2016-08-23:
As discussed in http://openrcforums.com/forum/viewtopic.php?f=94&t=9121 the replay speed is determined by examining the time stamps of up to the first 20 records of the log file entries and using the shortest time as the base replay rate.  The problem occurs because the first entry time stamp is delayed by the file open time and the time between the first and second entries may therefore be short.

I propose correcting by dropping the first record from the replay speed calculation and also setting the minimum replay rate to 100 ms.

closed 2016-08-27 by jtaylor2