opentx/opentx #3710: Inputting Telemetry through JR port on Taranis when PPM is enabled (D16 not enabled and no Bind)

marcdornan, 2016-08-21:
I am using ULRS for long range UHF RC and Mavlink control with Ardupilot. Currently I have found a solution that can convert Mavlink to Frsky on the ground side and do sensor polling required but with PPM enabled I understand you cannot input S Port telemetry. Can we get Telemetry fed in through the JR port without having D16 is selected a bind in place? 

This would free us from doing a relay with a FrSky X receiver. This could work for ULRS and Dragonlink V3 and allow just one teensy on the ground side.

kilrah, 2016-08-21:
>Currently I have found a solution that can convert Mavlink to Frsky on the ground side and do sensor polling

How?
The smart port polling is done by the module, so when it is off it won't poll...

geofrancis, 2016-09-28:
http://www.rcgroups.com/forums/showpost.php?p=35832289&postcount=7524

closed 2019-11-15 by stale[bot]