opentx/opentx #3714: How do I get Vario working with OpenTX 2.1.8 and D4R-II?

bitogre, 2016-08-23:
I do not know if this is a bug or an oversight regarding the FsSky Vario working with a D4R-II receiver.  I hooked up the Vario according to its manual at [http://www.frsky-rc.com/download/view.php?sort=&down=100&file=Manual-Variometer%20Sensor] with the Climb Rate DA output going into the A2 input of the D4R-II.

When I auto detected the sensors, it did create a "Alt" (ID 0021 0), "VSpd" (ID 0030 0), and "A2" (ID F103 25).  VSpd seems to be what I want without having to use the analog input (as the Vario manual suggests) but the "Vario" beep special functions does not seem to work as expected.  All I get is a regular repeating beep of constant frequency that does not change when climbing or diving.

I tried setting up "A2" based on the information in the manual.  Specifically, I changed units to "f/s" (also tried m/s), set precision to "0.00", Ratio to 6.2 (10.24/1.65), and offset of -3.12 (the value needed to have the reading give 0 with the aircraft stationary on the ground).  With this setting, it is clear that the reading is not signed like it should be as I get a huge positive number when the aircraft is descending (at least this part is definitely a bug).  In any case, I am not sure what to do next to tie this into the "Vario" beep special function.

What am I doing wrong?  Is this a limitation/bug/missing feature of OpenTX 2.1.8?  If not, where can I find documentation or other information on how to properly configure OpenTX for this?

3djc, 2016-08-23:
Have you setup Vspd as your vario source ?

fraternl, 2016-08-24:
this is a bugtracker and should NOT be used to ask or offer help

bitogre, 2016-08-25:
Ok, let me be more obvious about the bugs I am reporting:
1) Vario works automatically with X series receivers but not D series without having to explicitly set Vario source
2) A2 reading is not giving the proper negative reading when offset such that negative values should be possible.  Instead, I get ridiculously large numbers
3) Poor/Missing documentation/UI
3.A) It is hard to find where to set Vario source in the radio (I did eventually find it but that took too long due to being poorly documented and hidden on the page - have to scroll down a lot to find it)
3.B) Telemetry page [https://opentx.gitbooks.io/opentx-taranis-manual/content/telemetry.html] does not have a separate section for Vario.  The info in the manual on the Vario is in the same paragraph as other general information that is easily missed/overlooked.  And the information is minimal.  Should at a minimum include be a separate section with screen shots so people know what to look for.
3.C) None of the links at [https://github.com/opentx/opentx] provide links to forum/mailing list as a better place to ask/offer help making this the most obvious place to do so.  So, if not here, where should I go?  You indicate this is the wrong place for this but I still do not know where the right pace is.  If you do not make another place obvious then expect/accept people will use this for asking questions.  Or one of the links on that page (and several of the pages it links to) should be to where you want people to go to ask questions.

I work for a software company and we do use our bug tracking system to track any product issues, including UI and documentation issues.  If a feature is hard to use because it is poorly documented and/or hard to figure out, we consider it as serious a bug as if the feature was non-functional.  Why have code implementing a feature no one can figure out how to use.

3djc, 2016-08-25:
1) Vario as to be set with X series in the very same,way, at the very same place
2) Do not understand why you focus on A2 since it is Vspd that you should be looking at
3) I agree, it was SO difficult that googling "opentx forum" would get you, well to opentx forum. And also agree that the doc could be better, the only quetion remains what do you do about it ? It is free software done by volonteers for free, so if you think something should be better,  it is way more efficient to give a bit of your time to work on improving it than try to scream so that another volunteeer will do it for you ....

closed 2016-08-25 by 3djc

bsongis, 2016-08-25:
Agree, this project is a full time job and we can only spend our spare time on it (while others are flying with it)