opentx/opentx #3722: Companion V2.18N846  F4, RS not working,  LS wrong Channel reaction at Simulator

hrenz, 2016-08-25:
Transmitter Profil X9E
Hardware setting  F1,F2,F3,F4 as Potis  LS, RS as Sliders
source setting for test at mixer
LS=Ch6, RS= Ch7, F1=Ch8, F2=C9, F3=Ch10, F4=Ch11
Stimulator to Test
F1=Ch8, F2=Ch9,  F3=Ch10  works correct, ok
F4= Ch11,  RS= Ch7 no reaction
LS=CH6 no reaction, but woring movement at C11

[image]
[image]
[image]

kilrah, 2016-08-25:
Please stop opening these. The simulator does not support the X9E at this point and all aspects of that are considered on #2377.

closed 2016-08-25 by kilrah

hrenz, 2016-08-25:
ok, sorry was surrpriced that same things go, others not 

sorr