opentx/opentx #3723: Horus telemetry - Consumption reading incorrectly.

HorusChris, 2016-08-26:
Horus. OpenTx 2.2.ON350.
Using a FCS-150A Current sensor with X8R receiver.

Current readings match my Wattmeter within 1%.
Consumption readings however read at about 1/4 of the wattmeter reading.
This is then confirmed by recharging my battery.

So When Horus reads 20A. Wattmeter reads 20A.
When Horus shows consumption of 700mAh. Wattmeter shows just under 3000mAh
Recharging the lipo puts back Just over 3000mAh.

bsongis, 2016-08-26:
Strange I use myself the consumption with the Taranis and it is right.

HorusChris, 2016-08-26:
Yes It's right on my Taranis too.

kilrah, 2016-08-26:
> Yes It's right on my Taranis too.

With 2.2?

HorusChris, 2016-08-26:
No my Taranis is 2.1.0.

I'll try to get a telemetry log from Horus tomorrow. Current and Consumptione were not logging before.

bsongis, 2016-08-26:
Yes, excellent idea

crazynight78, 2016-08-27:
I have a similar problem with a frsky fcs-40a using 2.1.8, flight was over 50% out second was more like 25% seems to not display very low amp usage. I was flying a bixler with a lot of low throttle the first flight the second was more powered.

bsongis, 2016-08-27:
@crazynight78 Horus ? Taranis ?

bsongis, 2016-08-27:
I don't see any obvious issue in the code, I will wait for your logs

kilrah, 2016-08-27:
I've just set up a little test bench, need to charge a battery and I can run a test

HorusChris, 2016-08-27:
I may be some time with my results. Today is planned for family thing, not modelling. I got up to do this test and found no bind between Horus and my model. I've assumed this has been caused by flashing to FrTx and then back to OpenTx. During the flash to FrTx it did an IXJT upgrade.

HorusChris, 2016-08-27:
[Wots_Wot-2016-09-27.xlsx](https://github.com/opentx/opentx/files/440572/Wots_Wot-2016-09-27.xlsx)

HorusChris, 2016-08-27:
OK - seems that choosing the wrong FrTx file will switch the IXJT from FCC to LBT. So, having flashed to FrTx and back to fix that:

This mornings test is posted above. 
That shows a 1:20 run being held at around 20A with throttle (badly). 
Test stopped when Wattmeter was reading 400mAh.

HorusChris, 2016-08-27:
During the 20A period of my test, consumption should be increasing at about 5.55 mAh each second.

20*1000 = 20000mA and 20000/3600 = 5.55 mA/S

Log results show an increase of either 1 or 2 mA/s

kilrah, 2016-08-27:
I have just run a test discharging about 530mAh from a pack with 2 current sensors and 2 receivers, one bound to my X9E on 2.1.8 and the other to my Horus on current next branch, both matched, and my charger put about 540mAh back in.

So I'd presume you've got something configured wrong in your calculated sensor.

[image]

closed 2016-08-27 by kilrah

HorusChris, 2016-08-27:
OK thanks, that's good news. Now to work out what.
I assume my Current sensor must be set up correctly?

What is there that I can get wrong in my calculated one?
Set up like this:

Name = Cons
Type = Calculated
Formula = Consumpt
Sensor = Curr
Auto offset = blank
Positive = blank
Filter = checked
Persistent = blank
Logs = checked

kilrah, 2016-08-27:
Don't enable Filter.

HorusChris, 2016-08-27:
I'll give that a go when family allows, later today.