opentx/opentx #3733: Sensor Lost. Horus Open TX 2.2.0N350

KevinFairgrieve, 2016-08-29:
Beta Horus running Open TX 2.2.0N350.
FrSky X6R RX
FrSky GPS V2 Sensor

Flying with a rock solid link, but occasionally getting a "sensor lost" vocal alert.

Worked fine in this configuration using a Taranis running Open TX 2.1.8

I did video the flight on a hat cam and as soon as it is loaded to YouTube I will add the link so you can hear.

3djc, 2016-08-29:
Was it not telemetry lost ? (or are you running another language than english ?)

KevinFairgrieve, 2016-08-29:
No. Definitely "sensor lost".

English language used.

kilrah, 2016-08-29:
What sensors are connected?

KevinFairgrieve, 2016-08-29:
FrSky GPS V2 only

KevinFairgrieve, 2016-08-29:
I will check the connections and report back. Strange that it has worked on my Taranis for the last 18 months without issue.

kilrah, 2016-08-29:
That's because there was no such warning feature on 2.1.

KevinFairgrieve, 2016-08-29:
From about 1min 20sec in.

https://www.youtube.com/watch?v=Xse8yoLXgiw&feature=youtu.be

kilrah, 2016-08-29:
Got any logs?

bsongis, 2016-08-29:
It would be nice to see in the logs what is the refresh rate

bsongis, 2016-08-29:
And perhaps 15s is not optimized for GPS ...

KevinFairgrieve, 2016-08-30:
Sorry no logs.

Forgot to activate them during the days flights.

kilrah, 2016-08-30:
From the blinking '*' on the telemetry page refresh rate is 3Hz for speed and 1Hz for position/altitude.

I've tried simulating loss of GPS signal and that doesn't trigger the "sensor lost" alarm, info comes in regardless of validity. I can only simulate it by physically disconnecting the GPS, so that connection would still be the main suspect. Would be good to look at the telemetry setup page while checking the connection.

KevinFairgrieve, 2016-08-30:
Not sure this is going to help.

https://youtu.be/txKp8QBLpOU

Even changed to a different GPS Sensor.

When wiggling wires I saw no change to the telemetry settings.

May need to turn your volume up.

kilrah, 2016-08-30:
Can barely see a thing on the flickering screen... you should see a sensor go red when it happens. 
It looks like you might have 2 "GPS" sensors, that doesn't seem normal. Delete one (or both , and rediscover).

KevinFairgrieve, 2016-08-30:
The "*" do not stop on any sensor for any length of time.

Rediscovered, sometimes get 1 GPS others 2.

Hopefully better shorter video.

https://youtu.be/94ZNjoCenpg

projectkk2glider, 2016-08-30:
Looks like **Date** sensor gets no updates.

kilrah, 2016-08-31:
GPS sensor handling is just completely broken right now. Will be fixed after next test build.

KevinFairgrieve, 2016-09-01:
Thank you I shall wait the next rellease. Who knows I may even have a production radio......

bsongis, 2016-09-02:
Fixed

closed 2016-09-02 by bsongis

rondofpv, 2017-08-04:
Having the same issue after upgrading to the latest 2.2