opentx/opentx #3735: Companion 2.2.0N350 file save error in Linux

dwheeld, 2016-08-29:
[Companion 2.2 Downloaded companion-linux-2.2.0N350_amd64.deb nightly and also latest next compiled on Linux Ubuntu 16.04 LTS.] 

Steps to reproduce:

1. Run Companion 2.2 - Radio type Taranis
2. Create new model
3. Attempt to save new model as a new eepe (Taranis22airplane.eepe)

File is not saved and Companion displays following error message:
Cannot write file /home/me/desktop/taranis22files/taranis22airplanes.eepe:
Unknown error

kilrah, 2016-08-29:
Are you sure you're writing to a place you have permissions to?

dwheeld, 2016-08-29:
Yes. The exact same thing works fine with 2.1.8.

dwheeld, 2016-08-29:
Found why you can't duplicate the problem. I reported it as a Taranis problem instead of Horus. Here is how the issue should have been entered:

[Companion 2.2 Downloaded companion-linux-2.2.0N350_amd64.deb nightly and also latest next compiled on Linux Ubuntu 16.04 LTS.] 
Steps to reproduce:
1. Run Companion 2.2 - Radio type Horus
2. Create new model
3. Attempt to save new model as a new eepe (Horus22airplane.eepe)

File is not saved and Companion displays following error message:
Cannot write file /home/me/desktop/horus22files/horus22airplanes.eepe:
Unknown error

kilrah, 2016-08-30:
Horus model handling is not supported in companion at all yet.

bsongis, 2017-01-25:
Done in RC9

closed 2017-01-25 by bsongis