opentx/opentx #3738: nl.h.txt

fraternl, 2016-08-30:
Here's the updated translation file I made:

http://hmvc.eu/nl.h.22.txt

schwabe, 2016-08-30:
I took a quick look and it looks good. 

Some issues I noticed during a quick simu test:

(These I are just things I noticed. Not sure if they are intendend or not:)

String too long:

[image]

heli screen not translated: 

[image]

Dutch/English mix:

[image]

Not all special functions are translated:

[image]

Translation missing: 

[image]

I also added the modified file to the commit 75b74da2bd651414ca73bd6581ae1dd010f6dca6

fraternl, 2016-08-30:
Hi Schwabe,

Some of it is intended, but I agree it does look messy.
Our languages (Dutch, German) are very descriptive and take more words and letters than English.

The Dutch are mixing English words and expressions into their language much longer than Germans and therefore I chose to keep many things in English.

I did in fact a comparison with the German de.h.txt to check what was done there.

I will make some more corrections to the file...
Thanks

schwabe, 2016-08-30:
@fraternl No problem, I was just wondering.

The German translations is also a bit special since it translates almost everything and even changes the headings (e.g. from External Module to ======== Externes Modul ======). I don't use that one :)

fraternl, 2016-08-30:
The things you see are not in the nl.h.txt 
somehow the simulator uses something else or you are using an old copy....

I just wanted to correct some things, but these don't correspond with the screens you posted.

kilrah, 2016-08-31:
Merged without checking much, no time... but that way you can have a look once the next build is out.

closed 2016-08-31 by kilrah