opentx/opentx #3745: Question: Script Name Field in CUSTOM SCRIPT panel...

hymanroth, 2016-09-01:
I'm talking about the second field called 'Name' which is just below the "Script" field in the inputs section of the custom script panel.

Questions:

1 What's it's for? 
Surely the script already has a name (the name of the script file)
Does setting this value actually change anything?

2. How can I access the value of this field within Lua?
This might be useful as it's the only input that can take a text value

Thanks.

3djc, 2016-09-01:
This tool  is for tracking bugs and issue, which your question is neither. Please use a more suitable media, like this forum : http://www.rcgroups.com/forums/showthread.php?t=2180477

closed 2016-09-01 by 3djc

hymanroth, 2016-09-01:
OK @3djc if you're going to be like that then I'll change it to an ISSUE:

**Update to OpenTx Lua documentation to mention Script Name field in CUSTOM SCRIPT PANEL** 

Is that better?

schwabe, 2016-09-01:
Then it is still the wrong repo and should be issue in the lua doc repo (https://github.com/opentx/lua-reference-guide)

hymanroth, 2016-09-05:
Great idea @schwabe. If you answer my two questions, I'll be happy to update the lua docs.

schwabe, 2016-09-05:
I am by no mean a OpenTX lua export. But from looking at the source code that name field is only used in the UI screen and serves an informational value.

kilrah, 2016-09-05:
Exact, it's only for you to write e.g. what you're using it for.

hymanroth, 2016-09-05:
Ok thanks. It's a pity the value can't be read in Lua. It would be useful to have a text input field....