opentx/opentx #3746: timer can't start after reboot, opentx 2.1.8, Radio Taranis X9D Plus

xros, 2016-09-01:
timer can't start after reboot. I set a Time1, it is triggered by a logical switch, say `L8`. But every time I reboot the radio, I saw the time1 was `OFF` and have to re-map it to `L8`.

Do you have the same issue if tried?

bsongis, 2016-09-02:
Did someone confirm this issue? This seems really annoying!

bsongis, 2016-09-02:
I just tried to setup this (in OpenTX 2.2):
- a Logical Switch L8 with a condition THR > 0
- a Timer T1 with L8 as a trigger
Everything seems normal
I am not sure I understood correctly the issue.

xros, 2016-09-07:
@bsongis Yes, you understood it correctly. That's exactly the issue. Now we all have it.

bsongis, 2016-09-07:
But I had no problem, the timer starts correctly!

xros, 2016-09-07:
@bsongis My logical switch `L8`  is triggered by condition ( `a>x` 100 Throttle > -100 AND SG is up ), how about this?

kilrah, 2016-09-07:
Can't reproduce either, everything fine.

bsongis, 2016-09-08:
@xros perhaps you could attach here your model file?

xros, 2016-09-08:
@bsongis  Here's the model file
try set time1 starting at logic switch `L8` on a taranis x9d plus radio.

[x9d_plus_opentx_2.1.8_time1_debug.eepe.zip](https://github.com/opentx/opentx/files/460966/x9d_plus_opentx_2.1.8_time1_debug.eepe.zip)

kilrah, 2016-09-08:
That file has Timer 1 OFF, but when I set to L8 it doesn't go back to off in any way I can try.

Can you detail what you do? Is companion involved or are you doing everything on the radio? At what moment does it get lost?

xros, 2016-09-08:
@kilrah I tried this Time1 on the actual radio. You can try to set timer 1 at switch `L8` and check out whether it works on your radio. Then reboot, check whether the radio still map time 1 to switch `L8` or `OFF`. In my scenario, after rebooting the radio, time 1 lost mapping to `L8` and it's been `OFF`.

xros, 2016-09-14:
@kilrah If I mapped the trigger of timer 1, say, `SF Down`, after rebooting, it still lost trigger of the timer 1. I think this might be a general trigger lose. I saw the shuting down logo when turning the radio off, it's a _sleeping face_

kilrah, 2016-09-15:
Please answer my questions or we won't go anywhere.

xros, 2016-09-15:
> What happens if you change another parameter, does that get kept or lost?
Yes, it  loses again.
> What happens if you wait one minute between setting Timer1 start to L8 and turning the radio off?
It still loses after waiting for one minute.
> Do you see the "shutting down" logo when turning the radio off?
Yes I see

kilrah, 2016-09-16:
OK, then this suggests no changes at all are saved. I'd suspect a hardware issue on your radio, e.g. defective EEPROM chip.

bsongis, 2016-09-16:
It would be the second defective EEPROM within little time, I had another report from Wayne yesterday.

projectkk2glider, 2017-01-14:
Closing since is seem to affect only one radio - ie a hardware problem.

closed 2017-01-14 by projectkk2glider