opentx/opentx #3751: Crossfire protocol has no input to module

Webbsolution, 2016-09-02:
Using Beta hardware model 173 on firmware 351(RC)  - 

When you turn off the internal module and select the external module to use Crossfire ( and insert the crossfire module with appropriate connections) the RC input remains at "no signal" and no servo movement is seen -   - If you test with PPM as the protocol for the external module the signal is seen by the Module "8 ch PPM " and passes it to the servos.  So it would appear that crossfire protocol is broken in this rev.

bsongis, 2016-09-02:
Would you check something?

You flash OpenTX 2.2 on your Taranis (you will backup everything first as usual) and you test XF.
If it's not working, your module doesn't have the right XF version.

I will ask to Remo to comment about XF and OpenTX versions compatibility, I only tested non official versions ...

Webbsolution, 2016-09-02:
After the weekend - I have a 2 day F3K contest SAT and Sunday and I need to practice tomorrow :)

andrewbaum, 2016-09-02:
Just adding to this: on my Production Horus, before flashing OpenTX 2.2RC N351 and N352 - I installed my Crossfire and enabled external PPM in FrTX and the Xfire module showed it was receiving PPM input.  After loading OpenTX 2.2RC, the module shows 'No Input' regardless if the external module is set to PPM or CRSF (with int module disabled of course).  Xfire TX fw 1.40.

Webbsolution, 2016-09-02:
ok so thats 1.31/1.40 and 1.41 XF firmware that have issues - ill still be happy to check this on my taranis on 2.2 after the weekend but it sure looks like 2.2 seems to be the main issue.

bsongis, 2016-09-02:
It definitely needs another check. I will take care of it tonight.

bsongis, 2016-09-02:
[image]

closed 2016-09-02 by bsongis