opentx/opentx #3752: Internal and External modules can't be operated together with crossfire

Webbsolution, 2016-09-02:
Beta hardware unit 173 - firmware 351 

In the current rev if you select crossfire as the protocol the Internal module must be set to off. This precludes using both modules at the same time for a fall back band. This worked in open TX 2.1.7 with a taranis plus but seems to be restricted in the Horus. I understand the challenge is with a shared telemetry line which cannot be accessed by two modules at once. The crossfire module has an option to disable telemetry - my suggestion would be to put a " telemetry on" check box by the Crossfire protocol AND D8/D16/DSM etc - which can only be checked once with any combination of Crossfire and protocol X so either crossfire module gets the telemetry line OR Frsky gets the Telemetry line.   This gives a fixed wing/Quad/UAV pilot an option to run a Redundancy buss OR future facing RX with extensible ports as a satellite "in" option for a secondary band.

bsongis, 2016-09-02:
The XF protocol uses the S.PORT line for both transmitting (channels) and receiving (telemetry).
It's impossible to receive telemetry on the same line for the internal module. 

In 2.1 I forgot to do this check in the UI and the XF telemetry is not implemented, as all protocols have CRC all errors have been removed and you can see the telemetry of the internal module coming. But I am sure there are a lot of errors.

In 2.2 with the higher baudrate and the best usage of the bandwidth I don't think it's a good idea to allow to do that. The right solution would be to have different S.PORT lines on both internal and external modules ...

closed 2016-09-02 by bsongis

Webbsolution, 2016-09-02:
What if Telemetry is disabled on XF ? your issue is resolved and we could run both modules at once ?

kilrah, 2016-09-02:
If the XF gets an option to really disable its output, or if physically disconnected, yes

kilrah, 2016-09-02:
If the new XF version uses the SPORT line for communication both ways, no.

bsongis, 2016-09-02:
XF uses S.PORT for the channels as well

Webbsolution, 2016-09-02:
_ I'm trying to read between the lines here but it looks like what you are saying is that there is a single data line between both the internal and external modules and a single telemetry line or a single line for both channel and telemetry ? Are you saying that this is a hardware limitation ?? That there is no software approach to allow both modules to work at the same time? It was fine on Taranis with dozens of test flights... (using the Redundancy bus) -  If its truly a hardware limitation then speak clearly to this because its a devastating deficiency that essentially means even WITH a RB you cant have a dual band approach. I need to understand exactly where the limitation is and to what lengths it would take for a software approach to resolve (if possible)      

Laso if the answer is essentially yes - Crossfire + internal module cant ever be operating at the same time - then what is going on with PPM protocol on the external module? That is working right now too -

schwabe, 2016-09-02:
@Webbsolution Telemetry line is shared between external and internal module and XF uses that telemetry (instead of the PPM pin) for input and output. That is why PPM + internal work but not XF + internal

Webbsolution, 2016-09-02:
So if the Xfire module has a definite telemetry off option (1.41 does I have tested it) that would leave Frsky Telemetry to use the line and Xfire using nothing (that is an ok option for a lot of pilots using this as a back up band) So in that scenario what is the problem with adding BOTH modules as active options ? Why ois this such a big deal ??  This is a really critical issue because I am 90% sure that the radio will not support huge range and the L9R RX isnt an option for a lot of pilots who cant hang two draggy plastic paddles off their airframe.  In the Taranis this might have been an oversight - but - it worked so perfectly without a problem in the air - and it was the most redeeming feature when you look at adding a small redundancy bus. Without it the Horus simply cant be used by any advanced Glider or scale pilot who wants range like a competing radio provides....