opentx/opentx #3753: Turnigy 9X - invalid eeprom data

owenson, 2016-09-02:
Hi

OpenTX 9x 2.1.8
OpenTX Companion 2.1.8

Companion says invalid  eeprom data despite reading correctly from the atmel and opentx having settings setup on the eeprom (using the remote to configure).

Eeprom hex dump below:

[image]

Note, if I use the companion to program config into radio it says bad data on reboot.

owenson, 2016-09-02:
Tried this with master (companion and target) too. no luck.

owenson, 2016-09-02:
MainWindow::readEeprom(): using temp file:  "/tmp/1044-temp.hex" 
RLC decoding error! 
trying OpenTX for 9X board import... no
trying OpenTX for M128 / 9X board import... wrong size (2048/4096)
trying OpenTX for Gruvin9x board / 9X import... wrong size (2048/4096)
trying OpenTX for Sky9x board / 9X import... wrong size (2048/524288)
trying OpenTX for 9XR-PRO import... wrong size (2048/524288)
trying OpenTX for FrSky Taranis import... wrong size (2048/32768)
trying OpenTX for FrSky Taranis Plus import... wrong size (2048/32768)
trying OpenTX for FrSky Taranis X9E import... wrong size (2048/32768)
RLC decoding error! 
trying Gruvin9x import... no
trying Gruvin9x v4 import... wrong size
trying ersky9x import... wrong size
RLC decoding error! 
trying th9x import... no
RLC decoding error! 
trying er9x import... no
trying OpenTX for 9X board import... version 216 variant 0 ok

projectkk2glider, 2017-01-14:
Unless you can give us the problematic EEPROM file, we can't do anything about it. Reopen if needed.

closed 2017-01-14 by projectkk2glider