opentx/opentx #3755: More room in OpenTX 2.2 for the Dutch word for "warning"

fraternl, 2016-09-02:
When the radio starts it can give several warnings.
The Dutch word for "warning" is "Waarschuwing".
I can't really think of another word in this context.

The last letter (G) is not able to fit on the screen.
Can some extra room be made for this word???

bsongis, 2016-09-03:
And how would you see that?
I propose to finish the word directly on the case :)

mrpraline, 2016-09-03:
What is you use "Pas op!" instead of "Waarschuwing" ?

fraternl, 2016-09-03:
@bsongis funny man....
The font could be slightly smaller and it could move some pixels to the left...

But I didn't think of "Pas op!"
I like it even more.....
Thanks @mrpraline

fraternl, 2016-09-03:
A new file can be downloaden here:

http://hmvc.eu/nl.h.22.txt

fraternl, 2016-09-04:
On second thought "Pas op!" is not correct...
In Dutch you can say "Schakelaar Waarschuwing" like "Switch Warning" but "Schakelaar Pas op!" sounds awkward.  "Pas op! Schakelaar" is alright, but this would mean that it would change for other languages as well....

Please advice???
What changes are you willing to do?

kilrah, 2016-09-04:
Can't you just abbreviate it? We have to do it commonly in most other languages for some things, not a big issue...

lshems, 2016-09-04:
Why not use 'melding', as it actually isnt a warning, just a notification.

fraternl, 2016-09-05:
@kilrah  an abbreviaton for the word "warning" defeats its purpose and is very ugly. I don't want be the one to ruin @mhotar 's  work.
@lshems Given all the options I have it's the best way out I think. I will change it to that.

mhotar, 2016-09-05:
I do not use "Warning" word at all, I think the image says it already. So you have two lines to describe the problem. In Cz i have
"THROTTLE
CHECK
Throttle stick is not at zero"

fraternl, 2016-09-05:
en.h.txt
#define TR_WARNING             "WARNING"
#define TR_MIXWARNING          "Warning"

nl.h.txt
#define TR_WARNING             "MELDING"
#define TR_MIXWARNING          "Melding"

The #TR_WARNING is used together with another word on the opening screen of the model
There is insufficient room to place "WAARSCHUWING" there, the counterpart of "WARNING"

#TR_EEPROMWARN              "EEPROM"
#TR_STORAGE_WARNING    "STORAGE"
#TR_THROTTLEWARN          "THROTTLE"
#TR_SWITCHWARN              TR("SWITCH", "CONTROL")

SWITCH
WARNING

THROTTLE
WARNING

But it's OK now.... I will use the word "MELDING"

closed 2016-09-10 by fraternl