opentx/opentx #3761: Taranis not working with  BRCM20702 USB 3.0

dragon376, 2016-09-04:
I have been trying to get Taranis X9D + to work as a joystick on a Macbook pro late 2013 unsuccessfully.
It works on an older white Macbook and it works on PC, but on my most recent & powerful Mac, I can upload/download files to it, it does recognize it as a Joystick in System Information, but when running any Drone simulator or Joystick Doctor, I notice that the throttle doesn't work, and the other axis are sporadic, not covering the entire range even after calibration.

I have found out on this site:
https://github.com/opentx/opentx/wiki/Taranis-Bootloaders

that the following chipset is causing problems with Taranis.
Chipset: BRCM20702 USB 3.0
Product ID: 0x4500
Vendor: 0x0a5c  (Broadcom Corp.)

Since the same chipset works just fine when running windows, it is a driver software problem.
But it is reported that older version of boot loader was working. There might be a possible work around on your side?

dragon376, 2016-09-04:
BTW, This seems to only be a problem on El Capitan, not Yosemite....

kilrah, 2016-09-05:
I don't think it's something we can do much about. I had the same chipset on a 2014 Macbook Pro and it worked perfectly fine.

dragon376, 2016-09-05:
I just bought a $250 Taranis 9XD Plus mainly because Liftoff Games told me that controller was guarantied to work. LiftOff don't give a damn that it doesn't work, I'll see what FrySky says but I am expecting a deflection "we build the hardware, we don't have anything to do with the software", you are telling me it is not your problem, and Apple usually doesn't care about Niche Product...
I'll keep trying to find a driver assuming that this is the problem....
Is there anything else I am suppose to do after installing OpenTX to get the "Joystick" functionality?

kilrah, 2016-09-05:
Have you tried other USB cables? That's the number one reason for USB issues.

I remember something about needing a third-party driver to use joysticks on some mac OS versions.

dragon376, 2016-09-05:
I have tried 2 different USB cable, a 2.0 powered USB hub, a 3.0 non-powered USB hub,
That same cable, on that same Mac going through Parallel Desktop Win 10 works just fine.
It is a Driver issue on the mac side... Can you remember what third party driver? Cause I didn't find any doc or any forum that mentioned anything... I have read about dfu-utl, I have installed that and didn't do anything...

kilrah, 2016-09-05:
I believe USB Overdrive was an option

dragon376, 2016-09-05:
I found a really weird fix online that I didn't expect to work.
Creating a new User Account on the Mac and log on that account will make the Taranis Joystick work...
Do you have any insight as of what is the user dependent file or setting that would make it not work on one account but work on another account?
I am an SE, I have xCode installed on my this user account. I am wondering if that would make my main account have a debug version of some driver....

bsongis, 2016-09-05:
I guess your previous user is not in the right group and has no permissions to access such USB device

closed 2016-09-05 by bsongis

dragon376, 2016-09-05:
The previous user was my main account which is an admin, the user that works is a standard user...

jaydenlawson, 2018-02-10:
@dragon376 I have _exactly_ the same issue, with my 2013 Mac Pro 6,1. Have been trying everything for weeks - I even just bought an old USB 2.0 Hub as that was a suggested workaround because of some incompatibilities between USB 3 and USB 2 devices, even though it is supposed to be backwards compatible.

When I try and calibrate in Liftoff - it initially recognises the stick movement, but then doesn't recognise throttle, and all stick movement recognition is sporadic, jittery, and nowhere near smooth.  Basically - it is completely unusable. 

I'll try using the guest account now to install Steam and Liftoff.

jaydenlawson, 2018-02-10:
@dragon376 it also worked for me when creating a new profile. But I found the issue - it is USB Overdrive! Uninstalling it fixed the issue completely.