opentx/opentx #3766: Prevent (sometimes) the execution of LUA-script from SD-Card manager

fraternl, 2016-09-05:
The SD-Card manager can execute LUA-scripts.
When I was playing with it trying to discover the new features of OpenTX 2.2 I was able to execute a widget LUA-script after which the radio went in some loop that I only was able to stop by turning off the radio.

Maybe this feature (of executing a LUA-script) should be disabled on certain folders or only enabled on some folders...

bsongis, 2016-09-08:
Which script? There is a check to write at the beginning of the script to exit gracefully

projectkk2glider, 2016-09-10:
@fraternl please upload the script that freezes the radio. This should not happen.

kilrah, 2016-09-10:
Both lua widgets we distribute in the SD card image do it. It doesn't actually freeze the radio, but corrupts the display until reboot.

bsongis, 2016-09-10:
OK I will take care of this

fraternl, 2016-09-12:
Sorry for not reading this before.
I did make several model images in photoshop for the community

commit 4a88d347 (2016-09-13): Fixes #3766
closed 2016-09-13 by bsongis