opentx/opentx #3768: add vario tones as an RSSI indicator

Webbsolution, 2016-09-06:
When testing models for range with carbon gliders it is important to know the difference between RF shadows and generally low RSSI for extended periods of time while circling in lift at long range. I would really appreciate (as would others) if I could hear the impact in flight VS flipping toggles. It cant be that hard... most of it seems to be done.  

This works perfectly fine when the metric is climb or sink rate (with a vario) and its obvious bennefit cant be argued to soaring pilots - I think for RSSI tuning a RSSI trainer mode that sends ascending or descending tones out the speakers would show you in greater detail if your model shadowing once in a while or actually at the limit of its range without having to toggle every 2 seconds.

3djc, 2016-09-06:
Why would you toggle ? For your test, you could setup a repering annoucement (that can be turned on/off by a switch) every second if you so desire. You may also remove the unit to the rrsi to prevent having the repeting 'db'. You would then have a flow of rssi values.

Webbsolution, 2016-09-06:
Voice announcements are way too slow

projectkk2glider, 2016-09-10:
@Webbsolution I recommend you write a simple Lua mixer script that reads RSSI value and plays various tones depending on the value of RSSI.

Webbsolution, 2016-09-12:
There is a widget performing this function now -

closed 2016-09-12 by Webbsolution