opentx/opentx #3771: Horus WDT reboot reports

kilrah, 2016-09-06:
A few occurrences these few days:

- Reboot while editing settings on FM page https://opentx.rocket.chat/group/OpenTX_Horus_Testing?msg=xD38SZWjn3Nt6kMrN
- Reboot when creating a model (but user had messed with files in MODELS directory) http://www.rcgroups.com/forums/showpost.php?p=35654438&postcount=1218
- Reboot when copying a file in SD manager http://www.rcgroups.com/forums/showpost.php?p=35655849&postcount=1252

bsongis, 2016-09-07:
I tried to reproduce all of them, no luck

kilrah, 2016-09-07:
Me either... Must be a combination of things, or something random and unrelated to any of these... Would we try increasing the WDT timeout a little?

bsongis, 2016-09-07:
Why would the WDT fire when the user copies a file in the SD Manager? This task has the lowest priority, all other tasks should work as usual. I don't think increaing the timeout value would change anything.

bsongis, 2016-09-07:
I would vote for an SD card issue. Reading at the sdio_sd.c I see some race conditions, also I don't like the way the timeout decreases, without any wait, just an infinite loop. And at the end if the timeout expires there is no real test of the timeout at the end. Perhaps @projectkk2glider would you have a look at this file? I think I will propose a PR.

kilrah, 2016-09-11:
One more:
http://www.rcgroups.com/forums/showpost.php?p=35697212&postcount=2005

bsongis, 2016-09-11:
Model save when playing a sound?

projectkk2glider, 2016-09-11:
I am currently working on the SD card code to see if I can find any problem there. So far:
 * the timeout value of SDIO peripheral seems too long
 * the timeout code with infinite loop without any delay is also suspect.

I am installing some traces and will then begin to tune the wait parameters.

bsongis, 2016-09-11:
@projectkk2glider I have seen some differences in a file I downloaded from the web, I will commit it soon

projectkk2glider, 2016-09-11:
I thing the worst case to test is:
 * logs at 0.1s
 * constant audio prompts
 * model save

bsongis, 2016-09-11:
Indeed, I will do a test now

projectkk2glider, 2016-09-11:
How can a stuck SD card operation cause a WDT reset? I see no obvious way. 

The WDT is reset from the mixer thread and it does not use any IO functions. Only possibility is when mixers are paused, but that only happens during model load and certainly not while copying files.

I am now testing the worst case scenario (https://github.com/opentx/opentx/issues/3771#issuecomment-246177336) and see no problems....

bsongis, 2016-09-11:
Or an interrupt firing at full speed which would prevent the mixer thread from running

bsongis, 2016-09-11:
Or a problem on the stack

kilrah, 2016-09-11:
Freeze without reboot: https://opentx.rocket.chat/channel/OpenTX_General?msg=xFCpx43trJ5rWyqjQ

bsongis, 2016-09-11:
Impossible to reproduce here :(

kellybyrd, 2016-09-25:
I got this to happen today on 2.2.0 N358 Here is what I did:
- Created a model called "_TEMPLATE_". 
- Duplicated the template model.
- I noticed the "create category" option from the same popup dialog that "duplicate model"
- Created a category, called "Attic". My intent was to stick "_TEMPLATE_" in here.
- After category creation, UI puts me in the "Attic" category. I can't see any models. I can't find any way to get back to the top level. 
- When I use the "RTN" button from this Attic category, I got Emergency Mode.

kilrah, 2016-09-25:
Categories are flat, there is no "top level". You navigate between categories with PageUp/PageDn.

So the crash happens when exiting an empty category, right?

kellybyrd, 2016-09-25:
@kilrah: When I started creating models, there was not obvious category that I was in. I am calling this the "top level". As soon as I create a category, the UI switches me to that new category with no models in it and I cannot find a way out. when I try to exit with the RTN button got the crash.

3djc, 2016-09-25:
We will investigate the crash, what he as telling you is getting into the empty category is normal, you navigate to the other ones using PgUp and PgDown

kellybyrd, 2016-09-25:
Cool. Thanks for the PgUp/PgDn tip, I haven't gotten used to using those yet. I edited my comment above, besides the crash, I cannot find a way back the models that I created before creating any categories.

3djc, 2016-09-25:
You should have a models one by default I believe

kellybyrd, 2016-09-25:
Ok, I just spent some time in the model selector UI and now I understand. The default category is "All". I also see how to move models between categories. What remains is the bug I mentioned about exiting out of an empty category.

projectkk2glider, 2017-01-14:
Not much reports of this for a long time, closing.

closed 2017-01-14 by projectkk2glider

atshort, 2019-03-22:
I just had this happen. OpenTX 2.2.3 on Horus X10. Sitting on the bench, had been turned on for maybe three minutes. Not doing anything at the time this happened.

BladeScraper-Designs, 2019-05-20:
Same issue here.  Though not only does it enter emergency after turning off (and it reboots), I can hear the trim sound when holding the power button to power it down.
https://youtu.be/H27wtpO7-co