opentx/opentx #3777: Open TX Horus 2.2 ON352 - Dislay notes not shown

clivemartin, 2016-09-08:
When a new model selection is made from model select menu and activated, the 'text notes' are not shown on screen, however notes can be selected from long press on enter key for that newly selected model.  When transmitter is powered 'on' then notes are displayed for the last model selected.

clivemartin, 2016-09-09:
To be clear, it is the 'display checklist' (user generated text file) that is not displayed when a new model is selected from the model select menu. However with a model selected and the transmitter is powered OFF and then ON then the 'display checklist' is shown as first screen exactly the same as Taranis.

3djc, 2016-11-04:
Fixed in next nightly

PR #3986 (2016-11-03): 3djc/fix #3777
Delay the checks until after the main menu is launched to avoid the model notes to be overwritten even before appearing.

This fixes #3777
commit 00b23a42 (2016-11-03): 3djc/fix #3777 (#3986)

* Models notes where overwritten before even having a chance to be displayed

* Models notes where overwritten before even having a chance to be displayed (reverted from commit 0834894b61ed527c3c65d6356793a08ff6e3be6a)

* Delay the checks until after the main menu is launched to avoid the model notes to be overwritten even before appearing
closed 2016-11-04 by 3djc