opentx/opentx #3780: Need more "room" between options in widget set up using rotary encoder

Scott-Page, 2016-09-09:
I've had problems selecting some options using the rotary encoder when selecting widgets.  The problem is that the option I want fall between two detents in the rotary encoder and I can move back and forth and see the option I want but I have to try to  hold between the detents to select the option.

kilrah, 2016-09-09:
Any particular example? Never had the issue...

3djc, 2016-09-09:
I think it is a generic issue with the rotary encoder and extremely short lists

Scott-Page, 2016-09-10:
Specifically when choosing which widget I want in a location.  I saw somebody else said they had same thing in RCG.  It jumps to the next option too soon, so occasionally it jumps over one option. 

RSP
----------------------------------------------------------------------------------------------------------
When you were born, you cried and the world rejoiced. Live your life so that when you die, the world cries and you rejoice. - Cherokee

R. Scott Page
Photography & Broadcast Media Instructor
Hanford High School
450 Hanford St.
Richland, WA  99354
This email and any files transmitted with it are solely intended for the use of the addressee(s) and may contain information that is confidential and privileged. 
(509) 430-3921 (text)
http://www.hhsphotography.com

Scott-Page, 2016-09-10:
I figured out what's going on.  It's only when choosing the photo that it does this.  It's caused because it takes 1 to 2 seconds for the picture to scale and display.   Only way to prevent this would probably be to precache the image.  
Knowing WHAT is going on makes it easier to deal with.  I just need to wait until it loads before advancing to the next option.

RSP
----------------------------------------------------------------------------------------------------------
When you were born, you cried and the world rejoiced. Live your life so that when you die, the world cries and you rejoice. - Cherokee

R. Scott Page
Photography & Broadcast Media Instructor
Hanford High School
450 Hanford St.
Richland, WA  99354
This email and any files transmitted with it are solely intended for the use of the addressee(s) and may contain information that is confidential and privileged. 
(509) 430-3921 (text)
http://www.hhsphotography.com

kilrah, 2016-09-10:
Aha, then yes it's just a question of loading time, in both cases of widgets and images there is some significant processing to be done when loading.

Not much can be done without it becoming more of a pain somewhere else, even if we did background loading of the previous / next item you'd still have the issue on the "second idem" when navigating a list quickly... so I guess it's just a matter of "get used to it".

closed 2016-09-10 by kilrah

Scott-Page, 2016-09-10:
The other place I found difficulty with the rotary encoder is when choosing
the RGB values.  Two things.  1 - it is slow to respond.  2. a single
detent = 8 digits.  It would seem that one detent (slowly) would = 1
digit.  What this means is that nearly 90% of the RGB values are not
attainable.

RSP
----------------------------------------------------------------------------------------------------------
When you were born, you cried and the world rejoiced. Live your life so
that when you die, the world cries and you rejoice. - Cherokee

R. Scott Page
Photography & Broadcast Media Instructor
Hanford High School
450 Hanford St.
Richland, WA  99354
This email and any files transmitted with it are solely intended for the
use of the addressee(s) and may contain information that is confidential
and privileged.
(509) 430-3921 (cell)
http://www.hhsphotography.com

kilrah, 2016-09-11:
All normal. The slowness is due to the whole screen being rerendered with the new color. The steps are because the display uses RGB565 format, not 8-bit per color.