opentx/opentx #3788: better way of managing X series receivers

geofrancis, 2016-09-11:
In the model setup page when binding X series receivers they are given a number, when binding it would be good to be able to name the receiver for easier identification in the menu.

kilrah, 2016-09-11:
The actual identification method is a number, we can't just magically make it into a string.

I can't see how we could do anything better than you maintaining your own list (or stickers on the receivers?)

geofrancis, 2016-09-11:
just a text box attached to the number so it can be labelled manually would be enough.

schwabe, 2016-09-12:
I am wondering in what use case you need to identify the receivers (as to opposed to the model).

geofrancis, 2016-09-12:
It's mostly for testing purposes, I build a lot of machines and have receivers that I use for testing along with my own models and receivers in models that are being sold. 

At the moment it's it's easier to rebind each time rather than have a list of numbers or trying them all.

kilrah, 2016-09-12:
> At the moment it's it's easier to rebind each time rather than have a list of numbers or trying them all.  

That sounds easily solved by writing the number on the receiver as already suggested.

closed 2017-01-14 by projectkk2glider