opentx/opentx #3793: Proposal: Move antenna selector out of the model and put it into radio options

fraternl, 2016-09-12:
There's an option inside the model to select the external antenna.
This option is not saved as it is potentially dangerous for your RF-unit.

This means that if you want to use the external antenna always you have to enable it each time you switch models.

A radio-wide option would be just as safe and it minimizes the chance that you are using the internal antenna unintentionally. Some people might put directional antennas on there with a greater range.

kilrah, 2016-09-13:
It should actually be saved, if it isn't it's an overlook. I'd rather see the external antenna warning triggered at model load similarly to throttle warning etc.

fraternl, 2016-09-13:
Someone testing it on a Horus said it was saved. I only tested with the simulator. Not that I would test it on a Horus without an antenna.
Explicitly turning on the external antenna when you turn on the radio is much safer IMHO
I just ordered some dummy loads. I hope they fit and I can keep the cap on it.

kilrah, 2016-09-13:
> Explicitly turning on the external antenna when you turn on the radio is much safer IMHO

Excessively annoying though. A warning does the job.

fraternl, 2016-09-13:
Scratch the thing about the simulator...  it is saved.
I just tested it again. It was late....

If a warning is given each time you switch to a model with external antenna then it is at least covered.
Will you also be giving a warning when you switch from a model with external antenna to another one with external antenna??

mhotar, 2016-09-14:
I do not know if it is safe or even possible, but what about this test procedure: start module in low power (range) mode and test SWR for antennae presence ?

kilrah, 2016-09-14:
I doubt the SWR circuitry will work correctly at low power

projectkk2glider, 2016-09-14:
The IXJT module should take care of bad antenna internally. That is the only possible way and I assume FrSky did their home work on this. If not there will be a lot of unhappy folks with burnt RF section.

kilrah, 2016-09-15:
FrSky's word on it seems to be "it shouldn't be damaged but don't intentionally push your luck". 

From our side I'd suggest no more than a warning at model load if external antenna is selected.

fraternl, 2016-09-20:
If my suggestion of a radio wide session selection  could just be added upon the possibility of a model selection, it would mean it works best for everyone.
Those that occasionally or never use the external antenna rely on the model selection with its warning.

The ones that don't want to use the internal antenna could use the radio wide option before flying and after screwing it on.

fraternl, 2016-09-22:
Another addition to that would be a checkbox to remember the setting. The ones that keep the antenna attached in their case would probably prefer that. Those that take it off will be continuously remembered if they turn on their radio.

bigbobed, 2016-09-22:
Like the idea of a warning like throttle. Shouldn't be any more annoying than throttle or switch warnings.

kabturek, 2016-09-29:
Is the option for a antenna warning possible to add/enable when selecting the External module on something different than Horus ie Taranis? I have an external LRS module and that warning would allow me to sleep better ;-)

kilrah, 2016-09-29:
https://github.com/opentx/opentx/issues/1098

closed 2016-11-27 by fraternl