opentx/opentx #3795: horus lua widgets, input vars

natxopedreira, 2016-09-13:
Something its break on input values.

On the counter example widget, if you try to change the Option2 (defined as VALUE in code) the value does not increment up and down in a smooth way. Sunndenly it grows up to a wreid value.

And if i try to set a min/max range for VALUE type as per docs
{ "<name>", VALUE, <min>, <max>, <default> }

I get a Lua dissabled warning when boot, no errors show on debug panel

DeepnineRC, 2016-09-16:
Another error with input vars:

1.) when using negative values as default it breaks.

2.) When using multiple color pickers, the one not selected is missing the "R:" "G:" "B:" 

See video:
https://youtu.be/opwHzvXM69U

Lua option definition:
local options = {
  { "ArmCH", SOURCE, 1 },
  { "ArmOn", VALUE, -512 },
  { "ArmOnColor", COLOR, RED },
  { "ArmOff", VALUE, -1000 },
  { "ArmOffColor", COLOR, GREEN }
}

projectkk2glider, 2017-01-14:
This might be related #3882

projectkk2glider, 2017-02-19:
The issue with widget options was fixed in #4391. The order of options is not the same as order of parameters for inputs. I created https://github.com/opentx/lua-reference-guide/issues/30 to remind us. 

@natxopedreira read https://github.com/opentx/opentx/issues/3882#issuecomment-252239594

I will look at @DeepnineRC next.

RealTadango, 2017-02-21:
Looks like @DeepnineRC is missing min / max values in the option definition. That is new in 2.2 and needs to be added. Behaviour or the VALUE inputs is normal. 

The R G B labels are changing color when it is the active selection for ALL color inputs, not only the current selected one, that is a small gui bug i have noticed also.

commit b81c483b (2017-03-21): Fixes #3795 (#4643)

Multiple bugs found
closed 2017-03-21 by kilrah