opentx/opentx #3803: Minute timer on Horus

rcbebo82, 2016-09-14:
If i use the minute timer set to five minutes the system plays:
"four and zero"
"three and zero"
"two an zero"
"one an zero" 

It should play instead:
"four minutes"
"three minutes"
"two minutes"
"one minute"

projectkk2glider, 2016-09-14:
You most likely have a wrong sound pack. Opentx 2.0, 2.1 and 2.2 have different sound packs.

rcbebo82, 2016-09-15:
Well if have now used the original SD-CARD Horus nightly zip file and the error is also there!

3djc, 2016-09-15:
Appends in German it seems

rcbebo82, 2016-09-15:
I copied the english Sound files to the "de" Folder. Also used the csv inside the system folder and edited all paths and filenames to fit it. The error occurs also it seems there is something more wired up.

I now used the en folder and createt german sound files in it. Now all works as expected. 
Dont know how to get through the error. Anyone any idea?

kilrah, 2016-09-15:
> I now used the en folder and createt german sound files in it. Now all works as expected. 
Dont know how to get through the error. 

Sounds like you jsut did?

Pleas explain in detail what you tried, which language files, which language selected etc.

rcbebo82, 2016-09-15:
Yes it now works.
Voice Language: English
I took the csv file in the folder SOUNDS\EN\SYSTEM\en-US-taranis.csv and translated all english words to german. I also did some other changes to it to use with TTSAutomate 1.6.3.0. File attached here [](https://dl.dropboxusercontent.com/u/13465738/en-US-taranis%20%28aktuell%29.psv)
After i created all sound files i put the folder on the horus and overwrited all wav files.
The timer now works it Plays "vier Minuten", "drei Minuten", "zwei Minuten", "eins Minute".
So here we have the problem with the grammar it should play "eine Minute"

rcbebo82, 2016-09-15:
Mayby it has something to do with the en firmware i use on the horus?

kilrah, 2016-09-15:
That really doesn't help us to know what's needed to fix German...

rcbebo82, 2016-09-15:
I know .. i will do some further investigations with de sound files.

rcbebo82, 2016-09-15:
With the original de sound files from the sd card zip file and voice Deutsch selected the Horus plays on the minute timer:
4 und 0
3 und 0
2 und 0
1 minuten und 0
30
20
10
9
8
7
6
5
4
3
2
1

rcbebo82, 2016-09-15:
Seems the error is in tts_de.cpp

rcbebo82, 2016-09-15:
I think i found it tts_de.cpp:
Line 163: 
`PLAY_NUMBER(tmp, 0, 0);`

Unit is 0 istead of UNIT_HOURS

Line  170:
`PLAY_NUMBER(seconds, 0, 0);`

Unit is 0 instead of UNIT_MINUTES

Line 178:
`PLAY_NUMBER(seconds, 0, 0);`

Unit is 0 instead of UNIT_SECONDS

I hope this is this error. Im not so deep in OpenTx Code but that seems to make the difference to tts_en.cpp

3djc, 2016-09-15:
The reason is that plural sounds have not been genrated, will be fixed in next nightly

rcbebo82, 2016-09-15:
Just one question where do i find the allocation of the variable names to the file names?
 ` DE_PROMPT_COMMA = 102,`
` DE_PROMPT_UND,`
Here if find file 0102.wav is comma but und is not allocated to a filename.

3djc, 2016-09-15:
https://github.com/opentx/opentx/blob/next/radio/util/tts_de.py

rcbebo82, 2016-09-15:
Ah nice, but there are some translation errors!

3djc, 2016-09-15:
You are welcome to propose you changes, I habe Deutch in die Schule gelearned, aber ich habe alles vergessen !

rcbebo82, 2016-09-15:
Not that bad anyway 8-)
Deutch =  Deutsch, gelearned = gelernt

Made a pull request to that file #3809.

rcbebo82, 2016-09-16:
Installed nightly build 357 error is still there
4 und 0
3 und 0
2 und 0
1 Minuten und 0
30
20
10
9
8
7
6
5
4
3
2
1

3djc, 2016-09-16:
Indeed, 357 was released before those changes

rcbebo82, 2016-09-16:
Ah ok so it should be in tommorow nightly build?

3djc, 2016-09-16:
No, they are NOT build every night. It will be in N358, but I cannot tell when it is going to be available

rcbebo82, 2016-09-16:
Ah ok i understand just one last question: Doesent it make sense to name all sound files in all translations leaned to the english ones?
If anyone changes the voice and the file names are not the same Play track doesnt work anymore. Concret this happen on english flightmodes because the files in german have another name!

rcbebo82, 2016-09-23:
I had installed N358 now the timer works correct until one Minute is left.
It says "eins Minute und 0 Sekunden" it should say "eine Minute und 0 Sekunden".

rcbebo82, 2016-09-23:
In the "de-DE-taranis.csv" the line "SOUNDS/de/SYSTEM;0100.wav;100" must be replaced with "SOUNDS/de/SYSTEM;0100.wav;hundert". Dont know in which file this is done.

Problem is when the sound files are created the german voice says "einhundert" but this must be "hundert" because some other numbers are made up of two sound files eg. "zwei hundert", "drei hundert".

kilrah, 2016-09-23:
No, 200-300 etc are also single files.

rcbebo82, 2016-09-23:
In english yes but not in german and current nightly sd Card, mayby this is the Problem?

kilrah, 2016-09-23:
Don't know, need to look at the code in [TTS_de.cpp](https://github.com/opentx/opentx/blob/master/radio/src/translations/tts_de.cpp)

rcbebo82, 2016-09-23:
TTS_DE.cpp 
Line 152:
`
if (number >= 100) { 
     if (number >= 200) 
      PUSH_NUMBER_PROMPT(DE_PROMPT_NULL + number/100); 
     PUSH_NUMBER_PROMPT(DE_PROMPT_HUNDERT); 
     number %= 100; 
     if (number == 0) 
      number = -1; 
  } 
 PUSH_NUMBER_PROMPT(DE_PROMPT_NULL+number); 
`

If i understand this correctly PUSH_NUMBER_PROMPT plays the wav file. So this means for 200 firstly 0002.wav is played and then DE_PROMPT_HUNDERT=0100.wav

kilrah, 2016-09-23:
Looks like it's handled correctly then, and indeed just needs the spoken file to say "hundert".

See 
https://github.com/opentx/opentx/blob/master/radio/util/tts_de.py#L12

rcbebo82, 2016-09-23:
Correct that was my first idea! The problem is how the sound files are created. Currently 100 is spoken "einhundert" so the number 100 must be replaced with the text "hundert".
In the file tts_de.py i cant find anything according to that!

Another problem is especially with the number 100. If the upper problem is fixed 100 is spoken "hundert" but normally it should be spoken "einhundert". 
For this a new sound file is needed and the code needs a new if clause that matches all numbers between 100 and 199.

kilrah, 2016-09-23:
The link above points to the exact line that's affected. All numbes from 0 -100 get sent as is to the TTS engine, so 100 comes out as "einhundert". You would want to change that from 0-99, and a separate case for 100 that speaks the word "hundert", plus the new file. 

Then fix the code you posted above to take care of it.

rcbebo82, 2016-09-29:
Im currently working on the code.  What do these line of Code do? Especially what is mode in this context?
```
int8_t mode = MODE(att);
  if (mode > 0) {
#if defined(CPUARM)
    if (mode == 2) {
      number /= 10;
    }
#else
    // we assume that we are PREC1
#endif
    div_t qr = div(number, 10);
    if (qr.rem > 0) {
      PLAY_NUMBER(qr.quot, 0, 0);
      PUSH_NUMBER_PROMPT(DE_PROMPT_COMMA);
      if (mode==2 && qr.rem < 10)
        PUSH_NUMBER_PROMPT(DE_PROMPT_NULL);
      PLAY_NUMBER(qr.rem, unit, 0);
    }
    else {
      PLAY_NUMBER(qr.quot, unit, 0);
    }
    return;
  }
```

kilrah, 2016-09-29:
It's the precision. We don't store floats, but integers with shifted comma. So for example if precision = 2, a value of 1234 must be interpreted as 12.34.

rcbebo82, 2016-09-29:
Got it now the code makes sense to me, thx a lot kilrah!

rcbebo82, 2016-10-13:
Well it seem the orginial code in branch next for tts_de.cpp is already broken. On the simulator it wants to play 5535.wav, anyone any idea on that?

kilrah, 2016-10-13:
> On the simulator it wants to play 5535.wav

When doing what?

And yes it could very well be broken in other places - it hasn't been maintained for years AFAIK.

rcbebo82, 2016-10-13:
I created a global variable GV1 with value 100 and a special function which plays that value. It plays correctly 0100.wav and then 5535.wav. It is equal which value i configure for GV1 the last file played is every time 5535.wav.

kilrah, 2016-10-13:
What with the GVAR being 99? What with 101? 200? Should help you find what's wrong.

rcbebo82, 2016-10-13:
Next branch, without any code change. GV1 could be set to any number, it makes no change. Last file is anytime 5535.wav.
I think it makes sense to firstly find out why 5535.wav is played as last file. 
If this is corrected i can start working on the code which plays the numbers correctly in german.

3djc, 2016-10-13:
I'll investigate

kilrah, 2016-10-13:
Just did a test, it only does it on multiples of 100 for me. 
That's due to a missing test here:
https://github.com/opentx/opentx/blob/next/radio/src/translations/tts_de.cpp#L160

The code above sets number to -1 in case of the remainder being 0 after handling the 1000/100 readout, so this line tries to play -1, which translates to 65535 as it's converted to unsigned, and truncating that to 4 chars results in 5535.wav.
That line should only be executed if number > 0.

That at a quick glance would also make the 
```
if (number == 0)
      number = -1;
```
redundant.

3djc, 2016-10-13:
rcbebo82, this fixes next, you can use it to fix your branch maybe : 
https://github.com/opentx/opentx/commit/281c20bf361261fe354163938aedfb2021cc47f7

rcbebo82, 2016-10-13:
@3djc thx i will use this new code.

rcbebo82, 2016-11-09:
Fixed within #3936 !

closed 2016-11-09 by rcbebo82