opentx/opentx #3806: building on mac 10.11.6 fails during linking

iflymlnm, 2016-09-14:
Looks like multiple object files are being generated, 
Initially I had [this issue](https://github.com/opentx/opentx/issues/3699) but got past that,
any help/pointers is appreciated.

-S

duplicate symbol __ZTV10MainWindow in:
    CMakeFiles/companion.dir/moc_mainwindow.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZTS10MainWindow in:
    CMakeFiles/companion.dir/moc_mainwindow.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZTI10MainWindow in:
    CMakeFiles/companion.dir/moc_mainwindow.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN10MainWindow18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_mainwindow.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN16ModelsListWidget18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_modelslist.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN14ProgressWidget18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_progresswidget.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN11CopyProcess18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_process_copy.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN12FlashProcess18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_process_flash.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN11SyncProcess18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_process_sync.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN17MultiModelPrinter18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_multimodelprinter.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN12ModelPrinter18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_modelprinter.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o
duplicate symbol __ZN13WizardPrinter18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv in:
    CMakeFiles/companion.dir/moc_wizarddialog.cpp.o
    CMakeFiles/companion.dir/companion_automoc.cpp.o

iflymlnm, 2016-09-18:
I've tried last nite after RC5 and it builds fine, awesome!
You can close this issue.

schwabe, 2016-09-18:
No idea what happend there on your mac but good that it is working now! :)

closed 2016-09-18 by schwabe