opentx/opentx #3811: companion22_2.2.0N357_amd64.deb - Running on Linux Ubuntu 16.04 LTS – Will not automatically check for updates

dwheeld, 2016-09-15:
Steps to repoduce:

In Radio Profile:
Set Radio Type to FrSky Horus or FrSky Taranis.

In Application Settings:
Check box for - Use OpenTX firmware nightly builds.
Check box for - Automatic check for OpenTX firmware updates.
Check box for - Use Companion nightly builds.
Check box for - Automatic check for Companion updates.
Start Companion, or click on ‘Check for Updates’ button from main screen while Companion is running:

Shows error message:
Unable to check for updates.

Firmware can be downloaded by:
Start Companion.
Click on File.
Click on Downloaded.
Click on Check for updates, or Download FW.

3djc, 2016-09-15:
Uncheck non nightly, as there as not be a 2.2 release yet

dwheeld, 2016-09-15:
I still get the same results with any combination of boxes checked, or with none checked.

3djc, 2016-09-15:
Tested on debian, cannot reproduce, strange

3djc, 2016-09-16:
Could you check if the issue still exist for you ?

dwheeld, 2016-09-16:
Worked OK the first time I tried it, but now it is showing "Unable to check for updates" again. 

Also now when I got to File, Download, Check for updates or Download FW, it reports that it is unable to check for updates. These worked before.

kilrah, 2016-09-16:
Sounds like you don't have the 2 nightly checkboxes checked anymore.

dwheeld, 2016-09-16:
Correct. Working fine now. Thank you!

closed 2016-09-16 by projectkk2glider