opentx/opentx #3814: Enhancement: Show small preview image on model image selector

rcbebo82, 2016-09-16:
It would be nice if there could be a small preview image of the selected model image in the model image selector.

3djc, 2016-09-16:
Are you talking Horus ?

kilrah, 2016-09-16:
There is a preview in the SD card browser, Taranis has an "Assign image" command that can be used from there but that has yet to be ported to Horus.

rcbebo82, 2016-09-16:
Primarly Horus yes but this would be mayby also fine on X9E or X9D.

> There is a preview in the SD card browser, Taranis has an "Assign image" command that can be used from there but that has yet to be ported to Horus.

Ok but this means i have to look at the sd card browser and choose every image and open it.

It would be nice if i select the properly Image in model image selector wait a second and a preview would be displayed.

kilrah, 2016-09-16:
> Ok but this means i have to look at the sd card browser and choose every image and open it.

Precisely not, you see the preview straight while scrolling. Just go try it...

rcbebo82, 2016-09-16:
Got it, when this would be also in model image selector it would be very nice!

bsongis, 2016-09-16:
Yes, agree for this one, I will do it when things will calm down

kilrah, 2016-09-16:
I have a quick patch for the SD browser, but the whole thing would warrant further improvements (on all platforms, e.g. check we can only assign from the IMAGES folder). Do we wait and do it right, or shall I commit the patch?

bsongis, 2016-09-16:
Perhaps you open a branch, we are not forced to merge quickly, this is not a major issue

kilrah, 2016-09-16:
Reference didn't work... #3817

closed 2016-12-24 by kilrah