opentx/opentx #3815: Audio stuttering on Horus

projectkk2glider, 2016-09-16:
In the latest nightly 2.2.0N357. N356 was OK, the most likely culprit is 8a31c1d5551163d00b85d9576137619d73bb71dc 

The audio needs fixing, my proposal:
* add debug timer to the audio tak to see how often it is called and how long does one loop take
* implement DMA transfer on Horus
* other fixes that might be identified by the first step
* perhaps even have more audio buffers on Horus (we have more RAM)

projectkk2glider, 2016-09-16:
All reports say that the problem is only on startup with "welcome...." sound. Perhaps something is hogging the CPU time. at startup. It also seem dependent on the length of the power key press

> The Hello stutter seems to not be there if I hold the button down a bit longer - and is there if I very quickly hit the button

kilrah, 2016-09-16:
We load everything from SD at that moment, audio files are referenced, themes, widgets and all bitmaps are loaded, with all those things fighting for SD access we may well be doing a buffer underrun on the audio.

On my setup (custom audio files) it skips always at the same moment and the power button press duration doesn't change anything. BUT for me I always had occasional break on the startup file, it's just pretty much systematic now.

My files are higher bandwidth than stock so it's possible that while I had occasional stutter before and systematic now, others with lighter files now have occasional break when they had none before.

projectkk2glider, 2016-09-17:
First data is in. 

The startup sequence:
```
Audio int. : 3.510ms - 70ms
Audio dur. : 0.019ms - 70ms
 A. consume: 0.000ms - 1.481ms
 A push    : 0.000ms - 1.489ms
```

Later when opentx is running, no logging, no voice prompts:
```
Audio int. : 3.541ms - 4.457ms
Audio dur. : 0.019ms - 0.020ms
 A. consume: 0.000ms - 0.002ms
 A push    : 4294967ms - 0.000ms
```

Voice prompts:
```
Audio int. : 3.541ms - 19.998ms
Audio dur. : 0.019ms - 17.147ms
 A. consume: 0.000ms - 1.393ms
 A push    : 0.000ms - 1.939ms
```

Voice prompts and SD logs @0.1s:
```
Audio int. : 3.509ms - 19.999ms
Audio dur. : 0.019ms - 16.328ms
 A. consume: 0.000ms - 1.479ms
 A push    : 0.000ms - 1.490ms
```

First conclusions:
 * SPI writing to Audio chip is not a (big) problem, max 2ms, but the time could be used more useful by other task, so going to DMA is still good idea.
 * It is the reading from the SD card that is slowing down the Audio task, this can be seen by the *Audio dur.* which measures duration of `AudioTask::wakeup()`.
* the SD logs do not affect this

projectkk2glider, 2016-09-17:
I took a close up look at the Horus Audio driver. The conversion to DMA would be very difficult, if not impossible (two SPI speeds, BUSY signal, etc..).

Next interesting find: even with all SD card reads satisfied from the disk cache (so no actual SD IO is taking place), the audio task is taking a lot of time (cca 15ms):
```
>p dc
Disk Cache stats: reads: 9970, hits: 9770, misses: 200, hit rate: 97.9%
>p dc
Disk Cache stats: reads: 10069, hits: 9869, misses: 200, hit rate: 98.0%
>p dt
Audio int. : 3.510ms - 18.335ms
Audio dur. : 0.019ms - 15.890ms
 A. consume: 0.000ms - 1.478ms
 A push    : 0.000ms - 1.488ms
>p dc
Disk Cache stats: reads: 11240, hits: 11040, misses: 200, hit rate: 98.2%
```
I will install more debug timers to determine which part is time consuming.

projectkk2glider, 2016-09-18:
Results from ff6875fba504960f81d1850e61c202cb76125462

Only playing same files (from disk cache) no SD r/w:
```
Audio int. : 3.529ms - 14.001ms
Audio dur. : 0.020ms - 11.570ms
 consume   : 0.000ms - 1.478ms
 push      : 0.000ms - 1.486ms
 priority  : 0.001ms - 0.003ms
 tone      : 4294967ms - 0.000ms
 normal    : 0.010ms - 10.428ms
 vario     : 0.001ms - 0.012ms
 background: 4294967ms - 0.000ms
  mix open : 1.349ms - 10.196ms
  mix read : 0.003ms - 0.183ms
  mix resmp: 0.000ms - 0.100ms
  mix seek : 0.000ms - 0.001ms
   open1   : 1.308ms - 10.148ms
   read1   : 0.030ms - 0.154ms
   read2   : 0.004ms - 0.005ms
```

SD logs active:
````
>p dc
Disk Cache stats: w:162 r: 19011, h: 18784(98.8%), m: 227
>p dc
Disk Cache stats: w:167 r: 19236, h: 19009(98.8%), m: 227
>p dt
Audio int. : 3.536ms - 16.334ms
Audio dur. : 0.020ms - 13.507ms
 consume   : 0.000ms - 1.485ms
 push      : 0.000ms - 1.483ms
 priority  : 0.001ms - 0.003ms
 tone      : 4294967ms - 0.000ms
 normal    : 0.010ms - 12.018ms
 vario     : 0.001ms - 0.003ms
 background: 4294967ms - 0.000ms
  mix open : 1.390ms - 11.902ms
  mix read : 0.003ms - 4.233ms
  mix resmp: 0.000ms - 0.100ms
  mix seek : 0.000ms - 0.001ms
   open1   : 1.349ms - 11.845ms
   read1   : 0.031ms - 0.140ms
   read2   : 0.004ms - 0.005ms
````

Again the SD logs do not affect the Audio too much. The biggest time consumer is the `f_open()` function which can take up to 12ms (even 40ms in some cases where interaction with SD logs gets place). 

Strange thing is, all Audio reads are satisfied from the disk cache, so not SD IO reads are needed. But still the `f_open()` is so slow. There is a possibility that the read from the disk cache can be slow, because it is protected by the SD IO mutex, but that would only happen if for example SD logs were active.

projectkk2glider, 2016-09-27:
I am doing something big in a branch https://github.com/opentx/opentx/tree/projectkk2glider/issue_3815_audio_investigation The branch is dirty and needs a cleanup - I will prepare a new clean branch when I am satisfied. The main objectives are:
 * make Audio code more clean and readable, more object isolation
 * use new FatFS (fixes several bugs, easier handling of long filenames)
 * new more robust and simple audio buffer fifo
 * try too keep the audio buffers filled (ideally keep it full) this should reduce audio stuttering
 * simplify the audio thread - audio driver interface

projectkk2glider, 2016-10-25:
Looks like the new audio code did not cure the startup stutter issue. Examples of this are https://www.rcgroups.com/forums/showpost.php?p=36032884&postcount=4928 (https://www.youtube.com/watch?v=4oQP0-bwMGU&feature=youtu.be and https://www.youtube.com/watch?v=fStDSilV1FY&feature=youtu.be) The user is using N359.

When I analyzed the two samples is really looks like the ADC is starved of data shortly after the beginning of the wav file playing. I would guess that the first round of buffer filling goes smoothly, bu next time around the f_read() takes too long.

My idea is that audio thread gets starved, because the SD card is used for something else at the time (lots of reading going on at startup).

projectkk2glider, 2017-02-17:
Closing, audio seem to be working ok now, I moved experimental branch (which has some useful debugging tricks) to https://github.com/projectkk2glider/opentx/tree/projectkk2glider/issue_3815_audio_investigation

closed 2017-02-17 by projectkk2glider