opentx/opentx #3820: Resume after USB connection problems

projectkk2glider, 2016-09-17:
On Horus if USB=MASSSTORAGE, disconecting USB cable produces different results:
 * a GUI freeze
 * or a bad SD CARD message

Example trace, radio startup and USB cable connected:
```
register theme Default
register theme Darkblue
register layout Layout2x2
register layout Layout2x1
register layout Layout2x4
register layout Layout2P1
register layout Layout1x1
register widget ModelBmp
register widget Outputs
register widget Gauge
register widget Value
register widget Text
register widget Timer

Horus board started :)
>opentxInit
sdInit
sdMount
luaInit
register theme TBS
register widget Counter
register widget BattCheck
storageReadAll
loadFile(/RADIO/radio.bin)
loadFile(/MODELS/model1.bin)
disk_read bad alignment (0x100061ec)
load theme Default
disk_read bad alignment (0xd00f06bd)
disk_read bad alignment (0xd00f86b1)
opentxStart
�GC Use: 37616bytes
gps> $PUBX,40,VTG,0,0,0,0*5e
gps> $PUBX,40,GSV,0,0,0,0*59
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=294
gps> $PUBX,40,GSA,0,0,0,0*4e
gps> $PUBX,40,GLL,0,0,0,0*5c
eeprom write general
writeFile(/RADIO/radio.bin)
disk_write bad alignment (0x10003e50)
disk_write bad alignment (0x10003e50)
opentxClose
eeprom write general
writeFile(/RADIO/radio.bin)
disk_write bad alignment (0x10003e38)
disk_write bad alignment (0x10003e38)
sdDone
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=294
```

Disconnecting cable:
```
opentxResume
sdMount
storageReadAll
loadFile(/RADIO/radio.bin)
loadFile(/MODELS/model1.bin)
disk_read bad alignment (0x100061ec)
load theme Default
opentxStart
```

kilrah, 2016-09-17:
It was fine a couple of weeks ago...

projectkk2glider, 2016-09-17:
It might be just a bad USB cable that is corrupting the contents of the SD card. Will keep a close look.

projectkk2glider, 2016-09-18:
Another symptom:
* Radio apears to work normally, but subsequent USB connections do not work, I get:
```
Sep 18 21:44:23 mambo kernel: [215801.639768] usb 4-1.8: new full-speed USB device number 19 using ehci-pci
Sep 18 21:44:23 mambo kernel: [215801.711712] usb 4-1.8: device descriptor read/64, error -32
Sep 18 21:44:23 mambo kernel: [215801.887634] usb 4-1.8: device descriptor read/64, error -32
Sep 18 21:44:24 mambo kernel: [215802.207418] usb 4-1.8: new full-speed USB device number 21 using ehci-pci
Sep 18 21:44:24 mambo kernel: [215802.279385] usb 4-1.8: device descriptor read/64, error -32
Sep 18 21:44:24 mambo kernel: [215802.455306] usb 4-1.8: device descriptor read/64, error -32
Sep 18 21:44:24 mambo kernel: [215802.631160] usb 4-1.8: new full-speed USB device number 22 using ehci-pci
Sep 18 21:44:24 mambo kernel: [215802.703151] usb 4-1.8: device descriptor read/64, error -32
Sep 18 21:44:24 mambo kernel: [215802.879023] usb 4-1.8: device descriptor read/64, error -32
Sep 18 21:44:24 mambo kernel: [215803.054929] usb 4-1.8: new full-speed USB device number 23 using ehci-pci
Sep 18 21:44:25 mambo kernel: [215803.462684] usb 4-1.8: device not accepting address 23, error -32
Sep 18 21:44:25 mambo kernel: [215803.534661] usb 4-1.8: new full-speed USB device number 24 using ehci-pci
Sep 18 21:44:25 mambo kernel: [215803.942389] usb 4-1.8: device not accepting address 24, error -32
Sep 18 21:44:25 mambo kernel: [215803.942531] usb 4-1-port8: unable to enumerate USB device
```

projectkk2glider, 2016-10-07:
GUI freeze is explained in https://github.com/opentx/opentx/issues/3827#issuecomment-252325784

The USB connection problem also did not happen anymore since I re-seated the flat cable for USB board.

The third issue (no SD card) also does not happen anymore, must be the same fix as USB, since they share that flat cable.

projectkk2glider, 2016-10-08:
There is an SD card issue, here is how to reproduce:
 * connect USB cable, open disk on computer or whatever then disconnect the USB drive and disconect the cable
 * when the radio is in the switch warning screen (move switches in such position beforehand to get there) connect the USB cable again.

The result seen from computer:
```
ct  8 11:07:53 mambo kernel: [409891.014757] scsi 81:0:0:0: Direct-Access     FrSky    Horus   Radio    1.00 PQ: 0 ANSI: 2
Oct  8 11:07:53 mambo kernel: [409891.015185] sd 81:0:0:0: Attached scsi generic sg2 type 0
Oct  8 11:07:53 mambo kernel: [409891.015681] sd 81:0:0:0: [sdc] 3887104 512-byte logical blocks: (1.99 GB/1.85 GiB)
Oct  8 11:07:53 mambo kernel: [409891.016493] sd 81:0:0:0: [sdc] Write Protect is off
Oct  8 11:07:53 mambo kernel: [409891.016498] sd 81:0:0:0: [sdc] Mode Sense: 00 00 00 00
Oct  8 11:07:53 mambo kernel: [409891.016647] sd 81:0:0:0: [sdc] Asking for cache data failed
Oct  8 11:07:53 mambo kernel: [409891.016653] sd 81:0:0:0: [sdc] Assuming drive cache: write through
Oct  8 11:07:53 mambo kernel: [409891.023483]  sdc: sdc1
Oct  8 11:07:53 mambo kernel: [409891.024520] sd 81:0:0:0: [sdc] Attached SCSI removable disk
Oct  8 11:07:55 mambo kernel: [409892.365658] usb 1-2: USB disconnect, device number 67
Oct  8 11:07:55 mambo kernel: [409892.368614] scsi 81:0:0:0: rejecting I/O to offline device
Oct  8 11:07:55 mambo kernel: [409892.368620] scsi 81:0:0:0: [sdc] killing request
Oct  8 11:07:55 mambo kernel: [409892.368651] scsi 81:0:0:0: [sdc] FAILED Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
Oct  8 11:07:55 mambo kernel: [409892.368653] scsi 81:0:0:0: [sdc] CDB: 
Oct  8 11:07:55 mambo kernel: [409892.368655] Read(10): 28 00 00 00 00 10 00 00 08 00
Oct  8 11:07:55 mambo kernel: [409892.368663] blk_update_request: I/O error, dev sdc, sector 16
Oct  8 11:07:55 mambo kernel: [409892.368701] buffer_io_error: 124 callbacks suppressed
Oct  8 11:07:55 mambo kernel: [409892.368705] Buffer I/O error on dev sdc, logical block 2, async page read
Oct  8 11:07:55 mambo kernel: [409892.368733] Buffer I/O error on dev sdc, logical block 16, async page read
Oct  8 11:07:55 mambo kernel: [409892.368740] Buffer I/O error on dev sdc, logical block 16, async page read
Oct  8 11:07:55 mambo kernel: [409892.368745] Buffer I/O error on dev sdc, logical block 16, async page read
Oct  8 11:07:55 mambo kernel: [409892.368750] Buffer I/O error on dev sdc, logical block 2, async page read
Oct  8 11:07:55 mambo kernel: [409892.368754] Buffer I/O error on dev sdc, logical block 16, async page read
Oct  8 11:07:55 mambo kernel: [409892.368759] Buffer I/O error on dev sdc, logical block 8, async page read
Oct  8 11:07:55 mambo kernel: [409892.368764] Buffer I/O error on dev sdc, logical block 8, async page read
```

The result from CLI traces:
```
disk_write bad alignment (0x10003e40)
sdDone
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=460
opentxResume
sdMount
SD card info:
sectors: 3887104
type: 1
EraseGrSize: 1
EraseGrMul: 127
ManufacturerID: 2
storageReadAll
loadFile(/RADIO/radio.bin)
loadFile(/MODELS/model1.bin)
disk_read bad alignment (0x1000621c)
1000
load theme Default
opentxStart
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=460

<--------- THE POINT OF USB CONNECTION

Measure(storebackup) = 9601.0us
Status(ReadBlock)=3, s:16 c: 8
Status(ReadBlock)=3, s:16 c: 8
Status(ReadBlock)=3, s:16 c: 8
disk_read() multi-block failed, trying single block reads...
Status(ReadBlock)=3, s:16 c: 1
Status(ReadBlock)=3, s:16 c: 1
Status(ReadBlock)=3, s:16 c: 1
State=SD_TRANSFER_ERROR, c: 223587
Status(ReadBlock)=3, s:223587 c: 16
Status(ReadBlock)=3, s:223587 c: 16
disk_read() multi-block failed, trying single block reads...
Status(ReadBlock)=3, s:223587 c: 1
Status(ReadBlock)=3, s:223587 c: 1
Status(ReadBlock)=3, s:223587 c: 1
Status(ReadBlock)=3, s:16 c: 8
Status(ReadBlock)=3, s:16 c: 8
Status(ReadBlock)=3, s:16 c: 8
disk_read() multi-block failed, trying single block reads...
Status(ReadBlock)=3, s:16 c: 1
Status(ReadBlock)=3, s:16 c: 1
Status(ReadBlock)=3, s:16 c: 1

```

The error on PC could also show as (not sure if it has the same origin):
```
Oct  8 11:04:20 mambo kernel: [409677.225559] usb 1-2: USB disconnect, device number 59
Oct  8 11:04:29 mambo kernel: [409686.987500] usb 1-2: new full-speed USB device number 60 using xhci_hcd
Oct  8 11:04:29 mambo kernel: [409687.099429] usb 1-2: device descriptor read/64, error -71
Oct  8 11:04:30 mambo kernel: [409687.315310] usb 1-2: device descriptor read/64, error -71
Oct  8 11:04:30 mambo kernel: [409687.531173] usb 1-2: new full-speed USB device number 61 using xhci_hcd
Oct  8 11:04:30 mambo kernel: [409687.643105] usb 1-2: device descriptor read/64, error -71
Oct  8 11:04:30 mambo kernel: [409687.858983] usb 1-2: device descriptor read/64, error -71
Oct  8 11:04:30 mambo kernel: [409688.074849] usb 1-2: new full-speed USB device number 62 using xhci_hcd
Oct  8 11:04:30 mambo kernel: [409688.074997] usb 1-2: Device not responding to setup address.
Oct  8 11:04:31 mambo kernel: [409688.278873] usb 1-2: Device not responding to setup address.
Oct  8 11:04:31 mambo kernel: [409688.482598] usb 1-2: device not accepting address 62, error -71
Oct  8 11:04:31 mambo kernel: [409688.594557] usb 1-2: new full-speed USB device number 63 using xhci_hcd
```

projectkk2glider, 2017-01-14:
Closing for now, possibly fixed by #4125 and/or #4097.

closed 2017-01-14 by projectkk2glider