opentx/opentx #3821: Horus X12S N357 - LUA widget - key event

DeepnineRC, 2016-09-17:
Is there no key event on widget lua scripting on horus?

Or do i miss something?

kilrah, 2016-09-17:
Widgets have never been intended to be interactive... but it does seem like an overlook indeed, given there is thus no way to replicate what was possible with Taranis telemetry screens. We said "just make a full-page widget", but there's indeed no way to send it any keypresses.

Open to ideas on how to achieve that knowing there can be any number of widgets on a given page.

DeepnineRC, 2016-09-17:
> Open to ideas on how to achieve that knowing there can be any number of widgets on a given page.

Let the widget developer handle it. Send key events to all widgets on screen. I would make a setting field to my widgets "KeyEvent" On/Off.  Or make the setting field for all lua widget by default and by default set to off.
If you use different widgets that interfere with key events you can turn some off.

3djc, 2016-09-18:
It doesn't sound like a great idea, that would create widget that are incomptatible with each other. If you need to do something that requires imput, then I suggest doing like we do for S6R, a standalone script, not a widget

kilrah, 2016-09-18:
That's the worst way I could think of to deal with that...

For me you should be able to navigate between zones that have a widget loaded on the home page, select one, and then once one is selected it gets passed key events.

projectkk2glider, 2016-09-18:
>For me you should be able to navigate between zones that have a widget loaded on the home page, select one, and then once one is selected it gets passed key events.

I like this idea, and a long press on RTN would bring out the focus from the selected widget. When the widget is selected (ie receiving key events) it should have some graphical distinction (like special frame or background).

kilrah, 2016-09-18:
Yep, same frame around it as on the widget selector for example, dotted to choose the widget and solid once "in it".

I was going to suggest that a layout that has only a single zone could have key events always sent to the widget, but on second thought that would be too limiting since keys would still need to have their original function, those couldn't be used in the script. By "entering" into the zone one can send everything to the script except RTN LONG.

DeepnineRC, 2016-09-18:
Startup view like it is now is fine. No frame around the widget make i look clean.

Enter key mode: ENTER_SHORT ->Dotted frame (focus) appears on top left widget. 
Choosing frame to enter; With wheel left/right -> Dotted frame moves to next widget.
Enter the widget: ENTER_SHORT -> Dottet becomes to solid frame. Now you can use all keys except RTN_LONG. 

The key event on taranis was a bit dangerous when you pressed the wrong key, something happend in you lua script. With the widget/focus thing you can't accedently activate something in your script.

I suggest additional events:

1.) **evt_focus_over**: fires when the widget is chosen with the dotted frame, but not selected.
-> lua developer can make their own visual selection inside the widget

1.1.) evt_focus_loss: when moving to the next widget.
-> for stopping the the preselection.

2.)**evt_focus_active**: fires when widget is selected.
-> change to edit view or just change color.
2.1) **evt_focus_exit**: when leaving the widget selection. (RTN_LONG)
-> save settings in textfile, etc..

3.) **evt_focus_mode**: when the focus is visible to any widgets on screen.
3.1) **evt_focus_quit**: when return to normal view

raphaelcoeffic, 2017-01-06:
Is there anything new on this one? I think it would be highly valuable. Two applications I can think of: tune parameters of flight controllers, stabilised receivers, video transmitters, etc... One-time scripts can be pretty annoying to find and execute in a fast way.

dk7xe, 2017-02-27:
Since there are tune applications available for both [Betaflight](https://github.com/betaflight/betaflight-tx-lua-scripts) and [KissFC ](https://github.com/fedorcomander/kissfc-tx-lua-scripts) is is highly appreciated to get interactive LUA working on the Horus.

projectkk2glider, 2017-02-27:
Ideas from a chat:

> Given X10, I think we will have to improve a few things around menu handling and also LUA/widgets
If that format is going mainstream, I think will have issues linked to some of the things. I'm wondering if we should have a combination of flexible screen, but also so predefined ones maybe, where you can do different things (like interactive LUA) 

>One idea for so called *telemetry scripts* on X10/12 would be: If a widget uses the full screen, then it can use all the keys for interaction (except PgUp/Down) which are used to swithc to another screen. So any widget setup in a full screen mode could in effect function as a telemetry script on Taranis - ie allow user interaction without and weird side effects. Screens that have multiple widgets would not allow such interaction. 

>but I'd rather like the ability to select the active widget. The single widget in a 1x1 layout would be always selected "naturally"

dk7xe, 2017-02-27:
i really like that idea. if it is full screen allow key interaction except PgUp/Down.

Barbaryann, 2017-12-18:
Hi
Is there a chance to see a LUA widget to read lua script on horus ?
Thanks

lshems, 2017-12-18:
You can create an easy layer to simulate events based on stick it switch movement. I do that for my scripts. On my blog on rcgroups you can read how to load a telemetry script as a widget, and vice versa.

Although I do like the events from being accessable, I don't like at all being confined to some predefined workflow.

Just present a function to check if a widget is activated, but please leave the events visible on a general level for all widgets.

Barbaryann, 2017-12-18:
Thanks, do you have a link for your blog article about that ?
Thanks

lshems, 2017-12-18:
Search rcgroups for l Shems. That's me.

P.s. I do have a lot of examples where you want all widgets to respond to one event.

Three instances of a graphical log for three parameters, all reset on the exit button.

lshems, 2017-12-18:
Oh, and it is really simple to create a script run as one-time to simulate the taranis behaviour for multiple telemetry screens. Only disadvantage is that it runs only full screen and exclusive, so no Lua mixers for instance.

But who uses those in flight? They are advertised as not suitable for that.

Barbaryann, 2017-12-18:
I see yes. It is not very simple for a newbee in code than me :D

But i think it can't work for a Betaflight lua script or a tbs crossfir lua script where we can make configuration and many other thing... did you use it ?

lshems, 2017-12-18:
Thanks. Just suggestions. Not asking for anything. :) Do as you please.

kilrah, 2017-12-18:
> Just present a function to check if a widget is activated, but please leave the events visible on a general level for all widgets.
Really not very open to that. What would be your use case?

Any reason I could think of is precisely why you wouldn't want that - if someone starts looking at stuff even when not active then you'll have 2 doing that and they'll conflict with each other.

lshems, 2017-12-18:
Before getting the blame of asking and not being grateful for what I get.

lshems, 2017-12-18:
Syncing widgets is the most logical thing I can imagine supporting the idea of all widgets having access.

lshems, 2017-12-18:
@Barbaryann 
let's not clog this issue here. I have descrbed on my blog how to use script loaders to convert any type of script to any other type. Please comment there. I am sure you can already do whatever you want with the current setup. We'll discuss it up there to find a solution.

@kilrah 
On the subject of events for widgets: do whatever you like to limit them to one widget. If they are present to one widget, they are to all. Global variables are shared across widgets, so I'll just use those to signal the events to whatever script is running, even a mixer script.

But perhaps it would be nice if you tried to understand what I wrote, instead of jumping on it within a second, stating you don't like it. 

That you don't like me, is in the interest of nobody nor interests anybody.
That you are about to implement a crippled event system IS of interest to at east some people.

When you just having opened the possbility of easily configuring x widgets for measureing x different variables, and then dissalowing them from being ably to respond simultanuously to one key event, sounds crippling to me.

lshems, 2017-12-18:
@dk7xe 
you can use one time scripts just as you used telemetry scripts for this purpose.

lshems, 2017-12-18:
@DeepnineRC 

use this before your normal events handler, using two inputs as buttton replacements, being defnined in the options of the widget:

`local options,event = ...

if options.PLUSMINUS then

--plusMinus update Repeat progressive on time
if (options.lastEvent and ( getTime() - options.lastEvent > options.delay )) or (math.abs(options.PLUSMINUSValue or 100) < 20) then
  options.delay = math.max(10,(options.delay or 50) - 5) 
  options.lastEvent = getTime()
   
  if getValue(options.PLUSMINUS)/10.24  < -20 then
    event =  EVT_MINUS_FIRST or EVT_ROT_RIGHT
  elseif getValue(options.PLUSMINUS)/10.24  > 20 then
    event =  EVT_PLUS_FIRST or EVT_ROT_LEFT
  else
    options.delay = nil
    options.lastEvent = nil
  end
end
options.PLUSMINUSValue=getValue(options.PLUSMINUS)/10.24
end  
  
  
if options.ENTEREXIT then  
--set up of enterExit button simulation
--allow for dead band if continuous input is selected as source
if math.abs(options.ENTEREXITValue or 100) < 10 then 
  if getValue(options.ENTEREXIT)/10.24 < -20 then
    event =  EVT_EXIT_BREAK
  elseif getValue(options.ENTEREXIT)/10.24 > 20 then
    event =  EVT_ENTER_BREAK
  end
end
options.ENTEREXITValue=getValue(options.ENTEREXIT)/10.24
end

return event or 0`

jfrickmann, 2018-04-05:
I like projectkk2glider's suggestion above, to make a full screen widget more like the good old telemetry screen that we know and love. Alternatively, make an option to have a real telemetry screen in addition to widgets. I get frequent requests for a Horus version of my F3K program, but so far, I can only tell them to buy a Taranis,. It would be really nice to get this resolved.

By the way, my next generation program is a F3K and F5J package that you can find here: https://github.com/jfrickmann/SoarOTX 

Thanks for all of your hard work!!
Jesper

cprinos, 2018-10-26:
+1 on idea of events for a full screen widget.  Not being able to run https://github.com/jfrickmann/SoarOTX is the one thing stopping me from going QX7->X10

dgangola, 2019-03-03:
I wanted this feature on my X10S  badly enough to implement it.  My goal was to allow single-button access to features in standalone scripts. (Betaflight, in particular)  All that I wanted was a way to run a full-screen script, but it evolved from a script to a widget, and then I moved on to allowing small widgets to be interactive. 

Here’s what I have at this time:  Any Lua widget can be “interactive”, (receiving key events) if it so requests.  The widgets are interactive one at a time.  Request for interaction is done at load time, by returning a non-zero number in a “pages” variable in the array that contains the various entry points. (init, update, refresh, etc.)  This number also serves as a “page count”, the number of pages, (different views) that the widget can display.  Full-screen widgets (single zone) may have multiple pages.  A widget within a multi-zone layout can have only a single official page.

The radio keeps track of the pages.  A page number is sent to the widget’s refresh() routine, along with the key events.  The script is expected to display the page requested.  As the user moves through the screens and pages, (pressing page up/down) the current page is incremented, or decremented, and if either end of the pages (or the end of interactive zones on for a multi-zone page) is reached, the current screen will be incremented or decremented.

If a screen has no interactive widget, pressing page up/down will just move to the next or previous screen.  If the screen has an interactive widget, that widget will get the focus for key events.  If there are multiple interactive widgets on the page, the focus will shift forward or backward between them as the user presses page up/down.

 If the user is in the midst of a multiple page widget, and holds down the exit key, the current screen will change to the next or previous in the sequence, depending on the last direction travelled.  (page up or page down)  This is a short-cut to get past a lot of pages (Betaflight has 10) or many interactive widgets on a page.  For me, in real life use, I have one plain old telemetry page, with the Betaflight script.  When I’m done with Betaflight, I hold down exit, and I’m back to plain old telemetry.

The widgets do not receive the page up/down events, or long exit key events, except as an indication that they have received/lost focus.  EVT_PAGEDN_FIRST is sent on focus, and EVT_EXIT_LONG on focus lost.  The widget does not need to do anything with these events.  To save processing time, the widget should not redraw when receiving the focus lost notification.

Page numbers start with 1, with 0 being reserved for a “splash” page, selected when the user is setting up zones on a page.  This is to keep a script like the Betaflight script from trying to do telemetry during the widget selection process.

On a multi-zone page, the currently interactive widget has a border around it.  I’d like a lightly highlighted background, but drawing transparency doesn’t seem to work as expected.  About a 50% highlight is all that can be achieved, and that’s too much.
  
I’m planning to add a page map string to allow pages to be re-mapped in a more convenient order.  (For day to day use, I’d like the VTX page (page 8) in the Betaflight script to be first.  I could give a page map string of 812345679A, (or various shorthand schemes) to achieve this.  Unneeded pages could also be left out using the page map.  Page numbers are indicated by 1-9,A-F.

The source is on my github account at github.com/dgangola/opentx.  There is only one branch, drg/interactive_widgets, a branch from 2.2.3.  For compilation, set the option HORUS_INTERACTIVE_WIDGETS with CMake.  

My Betaflight widget is at github.com/dgangola/WIDGETS

Let me know what you think.

3djc, 2019-03-03:
We are quite definitively not satisfied with Horus widget/Lua handling. We have in the work rather deep changes to Horus underlying 'screen' handling, that should review all the handling of widgets and scripts. We havent reach LUA layer yet, we are still in main UI, but when we do, we will keep this thread in mind. I hope it can make it in 2.3, if not that should come quicly after that.

lshems, 2019-03-03:
Please keep it simple, without managing the user interaction over the widgets.

This can all be done within the widgets themselves, sharing the global Lua workspace.

Just propagating the events to the run function as done for telemetry scripts works fine.

A proper kill event to prevent page changes is all that is needed extra.

My opinion, well tested in several widgets with key simulation, that are used by my customers.

jfrickmann, 2019-03-03:
Great to hear that you are revisiting widgets and Lua! And I second lshem's comment about keeping it simple. As a matter of fact, it would be quite easy to implement loadable widgets in Lua entirely; there is no need for system support. I get frequent requests to make SoarOTX work with Horus, but for now, I have to tell them that they are SOL...

jfrickmann, 2019-04-13:
People keep bugging me to make a Horus version of SoarOTX. Is there any progress on this issue?
Thanks for all of your har work - much appreciated !!

lshems, 2019-04-13:
Not only SoarOTX.

Also Betaflight, Swiwin, GPS waypoint flying and many others. 

They are now very disappointed, or have a crippled widget where switches are use to mimic buttons.

P.S. that is proof that there is no need for any fancy logic; Just forward the events that are not used by the radio GUI, so at least **Plus/Minus and Enter/Exit.**

Long Enter is already useless due to the telemetry reset in a telemetry script. It pops up the model select on the X10.
Page up and Page down can be catched in a telemetry script, but are also used to switch between telemetry script pages, so don't bother there either.

I really have no clue how to program in C, otherwise I would have proposed a code change long ago. Spent my time trying to figure it out, but alas, don't get it.

kilrah, 2019-04-14:
This will come with the complete UI refactoring that's in progress.

jfrickmann, 2019-04-14:
Great, I look forward to that. Thanks again :-)

teckel12, 2019-05-03:
I believe in the meantime before complete UI refactoring:  Having the key events exposed if the widget is full-screen couldn't possibly cause any conflicts (as there's only a single widget).  Also, it wouldn't require any UI changes or controls to highlight the widget you wanted to send key events to.

I believe the main use cases are scripts designed to be full screen, like the BF script or my [INAV Lua Telemetry](https://github.com/iNavFlight/LuaTelemetry) script.  I've mostly got around the limitation by using the sticks to activate and navigate the config menu.  But, this limits the use to only when being unarmed and there's other limitations which don't exist on the Taranis.

It would seem this would also be a fairly simple thing to program. Basically, if full screen expose key events.  I'm not very familiar with the OpenTX codebase, and I don't have a development environment setup for building locally, but I may give it a whirl if the PR would be at least considered leading into the complete UI refactoring.  If due to the UI refactoring a PR wouldn't be considered, I'll just patiently wait for the UI refactoring like everyone else.

In the least, thanks for keeping this issue open and considering it in the UI refactoring.  It will be a very welcome feature and increase the usefulness of widgets.

lshems, 2019-05-05:
Please not limit to full size. Let the widget decide it is full size or not, and use it or not.

Starting two graphs at the same time with a button is a common scenario for me and others.

teckel12, 2019-05-05:
@lshems I would agree. But if there's some hesitation to add support due to possible confusion between multiple widgets, a full-screen limitation would make that argument invalid. The argument could then be made that key events should be exposed to all widgets. Baby steps...

lshems, 2019-05-06:
Yeah sure. But then they should count the number of widgets. 

Top bar, and screen.

Only if one widget visible, progress the events.

Or exclude top bar?

People writing scripts should be aware of how they work.

Users should trust scripts they run.

My interpretation of things, nothing new here.

lshems, 2020-02-22:
Not so stale.

closed 2020-11-20 by stale[bot]