opentx/opentx #3822: Horus X12S N357 - Curves resetting to random points

MyRCode, 2016-09-17:
I have had this issue happen on 3 different models.  

I have curves set for throttle hold, idle up, etc. for each of my 3 heli models.  I've been adjusting several curve points and and servo end points then exiting to the main screen and turning off the Horus.

When I turn the Horus back on, all the curves have been changed.  My 2 point flat-line curves have become 5 point diagonal curves, all of the untouched curves have become 5 point curves with random points.  No other sections of the model have changed.

This has happened on all 3 of my heli models and all after making multiple changes to multiple curves.

bsongis, 2016-09-17:
Is it *always* reproductible?

MyRCode, 2016-09-17:
It has happened on each of the 3 models after making changes to my setup.  I'm not sure what it is specifically that is causing the curves to randomize.  I can try to go through the same changes again in a few minutes...currently formatting my SD card.

MyRCode, 2016-09-17:
I'm not sure if this was also occurring on any of the previous firmwares.

MyRCode, 2016-09-17:
I just tried to reproduce it on N357 and could not.  I was going to flash back to N356 but I'll keep N357 for now and see if it happens again.  I've backed up my SD card onto my computer so should be able to go back to the old curves without having to enter everything manually.  Is there a way to take a screenshot on the Horus?

bsongis, 2016-09-18:
Still no, but I do believe you! It would be interesting to know if it is really fixed in N357, it's the first time we have such a report. Perhaps your models come from an older Horus alpha release and they were corrupted?

MyRCode, 2016-09-18:
I programmed the models from scratch on this Horus.   I was getting a windows error that the SD card had errors when I plugged in the Horus and windows chkdsk was unable to correct the error.   I swapped out the SD card for a new one and re-formatted so maybe that was related to this issue.

EricFPV, 2016-09-20:
I have same issue , flash to N358 , problem solved.

3djc, 2016-09-20:
If we can get another confirmation that it is fixed, it would be great

kilrah, 2016-09-20:
Yep, are you really sure? No way to consistently reproduce has been posted so far...

MyRCode, 2016-09-22:
Just had the curves reset again on N358.  The last thing that I did was change CV 1 from an 8 point curve with values of -93, -93, -93, -93, -93, 40, 70, 100 to -100 to 100.  

When I powered back on, all my curves were reset again.

bsongis, 2016-09-22:
Are you able to reproduce this problem? Only one curve was resetted, nothing else?

MyRCode, 2016-09-22:
I haven't been able to reproduce it so still not sure if that was exactly what did it. 

All the curves were reset again....I can't tell if they were reset to the same points again.  

I've reset everything back to the way it was before and backed up the SD card to my computer.

If it happens again, I'll copy the SD card to my computer before I make corrections and see if I can compare the two versions.

projectkk2glider, 2017-01-14:
Closing for now, reopen if problem re-appears in new versions.

closed 2017-01-14 by projectkk2glider

sloper-lloyd, 2019-09-29:
I have this issue on 2.2.2 and 2.2.4
X10S
Curves when edited affect other curves. Also when turn off and on 'some' curves have changed.