opentx/opentx #3823: Request:  Write protect on categories or models

MyRCode, 2016-09-17:
When selecting a new category using long press, delete category is one rotation away from select category.  It is very easy to accidentally delete a category.

iflymlnm, 2016-09-18:
+1 
I've created a few models and deleting the category wiped all my models, very very easy to accidentally delete potentially lots of models.

3djc, 2016-09-18:
The solution might be to have a confirmation box for deletion

MyRCode, 2016-09-18:
I believe there is a confirmation already,  but there is also a confirmation to select a different model so I was used to just hitting the short press twice.   

I've just had to take a mental pause in between the two passes just to verify I haven't accidentally scrolled the wheel.

closed 2016-10-18 by 3djc