opentx/opentx #3827: Wired output of all channels when Flight Data reset is choosen

rcbebo82, 2016-09-18:
Today i was at our flight field and found a curious error.
I flight my T-Rex 600 Nitro landed refueled it and choose "Reset flight data" on the horus after that the horus goes in in warning mode Throttle channel not zero and all Chanels get wirde output in all directions.

A video with the error: https://youtu.be/9WwtAGJWOyA

kilrah, 2016-09-18:
Are you using the internal RF or an external module?

bsongis, 2016-09-18:
No failsafe?

rcbebo82, 2016-09-18:
Internal RF with internal antenna. Receiver XSR connected by SBUS to the V-Stabi. Failsafe set to hold.

kilrah, 2016-09-18:
Is it reproductible? What happens if you turn the radio off?

rcbebo82, 2016-09-18:
Yes it is. Every time when a warning is active eg. Throttle not zero or any switch in a alarm position and flight data reset is done this happens. 
When warning is active and this is currently Happening if i press the power button jitterings stop immediatly after the horus is of the Receiver goes in failsafe mode.

rcbebo82, 2016-09-18:
It also happens on an other model with an X8R.

projectkk2glider, 2016-10-02:
Testing on Horus with 29e618146c673f605234a19210157c244c7d9cd4 I get another problem:
 * Initial throttle, switch and SD version warning - no problems
 * Flight reset
 * if I leave any warning displayed for more than 10s or so, I get total Menus thread freeze - no more display update. Radio still works and emergency shutdown also works. This does not happen every time, but at least in half of the tries.

projectkk2glider, 2016-10-07:
I have explanation for GUI freeze in warning display. When warning is in effect the `getADC()` function gets called from two threads: main and mixer. Downstream it uses the `adcRead()` which is not reentrant! A mutex should solve this, hold on...

projectkk2glider, 2016-10-07:
@rcbebo82 can you test my fix if it solves your issue?

rcbebo82, 2016-10-07:
Sure, i will do this tomorrow.

rcbebo82, 2016-10-08:
I cant clone your branch projectkk2glider/issue_3827_adc_mutex because i dont see it. How can i clone it?

projectkk2glider, 2016-10-08:
Just checkout my branch.  You need to pull the repository before you will see the new branch.

rcbebo82, 2016-10-08:
I dont understand that. I cloned your branch opentx but there is no branch "issue_3827_adc_mutex".
[image]

kilrah, 2016-10-08:
You're confusing cloning and checking out. 

We're talking of checking out a branch called projectkk2glider/issue_3827_adc_mutex in the main OpenTX repo, NOT in projectkk2glider's fork of OpenTX. That one's probably been disused since he was added as a member of OpenTX.

rcbebo82, 2016-10-08:
Ah got it! Thx. Im soon back with Information if the problem has gone away!

rcbebo82, 2016-10-08:
Problem fixed with your branch!

projectkk2glider, 2016-10-08:
Thanks for confirmation.

closed 2016-10-14 by kilrah